FreeRTOS/FreeRTOS-Kernel #594: [Feature Request] Enable completely disabling the heap through CMake

CrustyAuklet, 2022-11-30:
**Is your feature request related to a problem? Please describe.**
I am using FreeRTOS on a very small MCU where we want no heap implementation. This is normally done by
setting `configSUPPORT_DYNAMIC_ALLOCATION` to 0. But I don't see a way to disable the heap via CMake.

**Describe the solution you'd like**
I would like to be able to set the CMake cache variable `FREERTOS_HEAP` to a "falsey" string and have no heap file added to
the project. Strings that CMake considers false are described [here](https://cmake.org/cmake/help/latest/manual/cmake-generator-expressions.7.html#genex:BOOL)

I think this is a pretty simple fix, to change the generator expression to
```
# If FREERTOS_HEAP is digit between 1 .. 5 - it is heap number, otherwise - it is path to custom heap source file
# Disable heap if set to an empty string, 0, FALSE, OFF, N, NO, IGNORE, NOTFOUND, or ends with -NOTFOUND
$<IF:$<BOOL:${FREERTOS_HEAP}>,$<IF:$<BOOL:$<FILTER:${FREERTOS_HEAP},EXCLUDE,^[1-5]$>>,${FREERTOS_HEAP},portable/MemMang/heap_${FREERTOS_HEAP}.c>,>
```
I have this in a branch on my fork [here](https://github.com/CrustyAuklet/FreeRTOS-Kernel/blob/cmake-allow-no-heap/CMakeLists.txt)

**Describe alternatives you've considered**
Looking at the CMakeLists.txt in this repo I don't see a lot of other options. I tried setting `FREERTOS_HEAP` to an empty string and it tried to add `portable/MemMang/heap_.c` to the project. I tried just using a heap file but then I get the error

```
#error This file must not be used if configSUPPORT_DYNAMIC_ALLOCATION is 0
```

**How many devices will this feature impact?**
thousands

**What are your project timelines?**
Working on an MVP shipping date in January

**Additional context**
None.

If you have the same (or similar) feature request, please upvote this issue with thumbs up 👍
and use the comments section to provide answers to the questions above.

jasonpcarroll, 2022-11-30:
Hi @CrustyAuklet,
This definitely seems like an oversight. Would probably be easier to just have a bool for it, like -FREERTOS_DO_NOT_INCLUDE_HEAP or something along those lines that can just be added to your higher level Cmakelists or passed on the command line -DFREERTOS_DO_NOT_INCLUDE_HEAP. I'll make a PR for it on my fork for you to try.

jasonpcarroll, 2022-11-30:
Could you try my fork and see if that fixes your issue? I haven't actually tested it yet, but it should work if you add -DFREERTOS_DO_NOT_INCLUDE_HEAP to the command line for cmake. Alternatively defining FREERTOS_DO_NOT_INCLUDE_HEAP in a higher level CMakeLists should also work.

CrustyAuklet, 2022-12-02:
That works too, and I think I like a little better. I wasn't sure if it was OK to add another variable but it is a bit more explicit that way.

closed 2023-12-06 by chinglee-iot