FreeRTOS/FreeRTOS-Kernel #599: [BUG] Cortex-M33 port may not reset MSP as expected

stevew817, 2022-12-08:
**Describe the bug**
Cortex-M33 may not always be resetting the MSP as expected during `vStartFirstTask` if there are pending IRQs before enabling the kernel.

I believe the Cortex-M33 may require an ISB between the setting of the MSP and the enabling of interrupts here: https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/portable/GCC/ARM_CM33_NTZ/non_secure/portasm.c#L188

**Target**
- Development board: Silicon Labs EFR32FG23
- Instruction Set Architecture: ARMv8-m
- IDE and version: IAR 9.20.14
- Toolchain and version: IAR 9.20.14

**Host**
- Host OS: Windows
- Version: Windows 11

**To Reproduce**
Prime an IRQ before starting the scheduler, and observe that the main stack has traces of the call chain from `main` into `vTaskStartScheduler`, followed by `xPortStartScheduler` and `vStartFirstTask`.

**Expected behavior**
After the call to `vStartFirstTask`, the MSP points at the original initial stack pointer from the VTOR.

**Additional context**
A forum user indicates that it might be architecturally required to have an ISB following the change of the currently-in-use stack pointer with `MSR` (ref https://www.freertos.org/FreeRTOS_Support_Forum_Archive/March_2017/freertos_ISB_isntructions_on_Cortex-M4_port_6fc12825j.html), but I haven't found specific language around that in the ARM documentation yet, other than generic language around manual synchronisation when changing control and/or memory map defining registers.

kstribrnAmzn, 2022-12-08:
Thanks for taking the time to report this with such detail! I especially appreciate your meticulousness as I haven't worked much with this port so I'll be reaching out to those on my team who have. I'll circle back here in the coming days with whatever information I come across.

urutva, 2022-12-14:
@stevew817 Thanks for reporting the issue with such detail! I observed a discrepancy in the toolchain mentioned (IAR) `Toolchain and version: IAR 9.20.14` and the code referred (GCC) `https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/portable/GCC/ARM_CM33_NTZ/non_secure/portasm.c#L188`. Can you please check this?

Also, it'd be helpful if you could share the disassembly of `main`, `ISR` and `vStartFirstTask` that causes the issue reported.

Thanks

RichardBarry, 2022-12-14:
My understanding is this is caused by a missing barrier instruction in the FreeRTOS kernel code on the line highlighted in the original bug report.  A barrier between changing the stack pointer and enabling interrupts will ensure the new stack pointer is in effect before any pending interrupts execute.

stevew817, 2022-12-14:
@RichardBarry That is my understanding as well.

@urutva You are correct in that I referenced IAR and linked to the GCC-port. The compiler doesn't matter in this case though (and so the same issue would be present on GCC) since the issue is in `vStartFirstTask` which is handwritten assembly code. So the same fix would need to be applied for all ports of ARMv8-m.

RichardBarry, 2022-12-15:
Excellent info - thanks for the explanation.  Maybe I didn't get it wrong in the code after all :o)

closed 2022-12-30 by alfred2g