FreeRTOS/FreeRTOS-Kernel #609: [Feature Request] Tags need to probably refer to commits/branches IN the repo here...

madscientist42, 2023-01-11:
**Is your feature request related to a problem? Please describe.**
Anything past V10.4.3 can't be pulled by tag with this repository.  "This commit does not belong to any branch on this repository, and may belong to a fork outside of the repository." is what GitHub tells you when you go to the releases tab and then try to pull up the commit for the tag.  No bueno.  

**Describe the solution you'd like**
Well, tags SHOULD correspond to the commit, as it exists in the repo.  **_THIS_** repo.  If it doesn't there's no good use to have versioning.  I should be able to refer to the tag itself and pull it instead of stashing a Githash.  It's one of the functions and features of tags.

**Describe alternatives you've considered**
What alternatives...?  You have messed up your versioning info.  Needs to be fixed.  But it's not a bug, per se.  (Feel free to move this over if you think it's more a bug...)

**How many devices will this feature impact?**
Anyone that actually tries to use tags like they're supposed to.  Oh, say, Yocto, etc.  Right now I'm trying to do a cleaner integration of FreeRTOS build under Yocto than the current meta-freertos (Not bagging on them- I just can't use the stuff as-is in my project for a Fortune 500 level client.) for a Xilinx target that can build either for a Cortex-R5 or a Microblaze/RISC-V core on the FPGA dependent on the boards being built out for.  Git hashes are clumsy.  Version tags?  **CLEAN**.  Can be against the packaging version, etc.

kstribrnAmzn, 2023-01-14:
Thanks for opening the issue @madscientist42. Obviously this is troubling. Can you walk me through what you're trying to do?

I was able to both clone at the tag and checkout a tag with the following commands...
```
//Clone at tag V10.5.1
git clone -b V10.5.1 git@github.com:FreeRTOS/FreeRTOS-Kernel.git

//Checkout tag V10.5.0 on branch
git fetch
git checkout tags/V10.5.0 -b anotherTag
```

If `git fetch` doesn't work, can you try `git fetch --tags` to update the tags index? See https://github.com/grafana/grafana/issues/45642

archigup, 2023-01-19:
Hey, the commits will be pulled when you get the tag. Commits referenced by a tag are part of the repo, not just branches.
The tagged commits are the release commits, but are not in the `main` (development) branch. Version tags work like normal.

closed 2023-01-19 by archigup