FreeRTOS/FreeRTOS-Kernel #615: Why the heap_4.c buffer is tagged as `PRIVILEGED_DATA`?

g-berthiaume, 2023-01-19:
Hi,
I'm porting the FreeRTOS Kernel to a TI RM46 (cortex r4) with the MPU activated.
I was wondering why the heap_4.c buffer is [tagged as `PRIVILEGED_DATA`](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/8592fd23f4fbc1a2fa821a9b89b61ef1f5652a09/portable/MemMang/heap_4.c#L92)?

To my understanding, if the heap is privileged, it implies that all the tasks' stack are also privileged.
It seems like it would defeat the purpose of the MPU.

paulbartell, 2023-01-19:
@g-berthiaume

With the MPU enabled, it is most common to statically allocated task stacks at compile / link time.

In targets implementing ARM PMSAv7 (such as the Cortex-R4), MPU regions may overlap with the higher numbered region taking precedence over lower numbered regions. This would in theory allow task stacks to be allocated from the system heap (located in the portPRIVILEGED_RAM_REGION region) with the smaller stack MPU region (portSTACK_REGION==) taking precedence.

Note that ARM PMSAv8 is very different in terms of size and alignment requirements. In some ways it is more flexible however it does now allow for overlapping regions other than the optional background region / "system memory map" controlled by the PRIVDEFENA field in MPU_CTRL.

The current heap_4.c implementation does not support arbitrary alignment of memory regions. To make this work, you must enable configSTACK_ALLOCATION_FROM_SEPARATE_HEAP and implement the pvPortMallocStack and vPortFreeStack functions defined in [include/portable.h](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/include/portable.h) to allocate task stacks in a way that complies with the MPU requirements namely:

### Alignment in ARM PMSAv7
region_base_addr = ( ADDR << 5 ) where ADDR is a 27-bit field in the MPU_RBAR register.
The effectively means that MPU regions must start on a **32-byte** aligned address.

### Region Size in ARM PMSAv7
region_size = 2^(SIZE+1) where SIZE is the 5-bit field in the MPU_RASR register.

Values of SIZE less than 4 are reserved, so region size is a power of two between 32 (2^5, SIZE=4) and (2^33, SIZE=32) of one of the following values:

| SIZE | region_size |
|------|------------:|
| 4    |        32 B |
| 5    |        64 B |
| 6    |       128 B |
| 7    |       256 B |
| 8    |       512 B |
| 9    |       1 KiB |
| 10   |       2 KiB |
| 11   |       4 KiB |
| 12   |       8 KiB |
| 13   |      16 KiB |
| 14   |      32 KiB |
| 15   |      64 KiB |
| 16   |     128 KiB |
| 17   |     256 KiB |
| 18   |     512 KiB |
| 19   |       1 MiB |
| 20   |       2 MiB |
| 21   |       4 MiB |
| 22   |       8 MiB |
| 23   |      16 MiB |
| 24   |      32 MiB |
| 25   |      64 MiB |
| 26   |     128 MiB |
| 27   |     256 MiB |
| 28   |     512 MiB |
| 29   |       1 GiB |
| 30   |       2 GiB |
| 31   |       4 GiB |

Any allocation metadata should be located outside of stack regions to prevent against a variety of privilege escalation attacks. 

In your pvPortMallocStack implementation, you can probably just allocate your desired region size + 32 from the system heap and align the resulting pointer to 32-bytes. Then do the reverse in your vPortFreeStack impementation.

paulbartell, 2023-03-09:
@g-berthiaume 

The supported configuration is to always statically allocate the stack for unprivileged tasks.

We will consider allowing for dynamically allocated unprivileged task stacks in a future release.

closed 2023-03-09 by paulbartell

g-berthiaume, 2023-03-10:
Thanks for the update