FreeRTOS/FreeRTOS-Kernel #617: [Feature Request] Replace sprintf with snprintf

Hadatko, 2023-01-26:
**Is your feature request related to a problem? Please describe.**
Hello, newer compilers are starting to complain related to usage of sprintf function (clang returning warning about deprecation). Their are suggesting to replace that call with snprintf. May you consider align your code?

**Describe the solution you'd like**
Replace sprintf with snprintf

**Describe alternatives you've considered**
-

**How many devices will this feature impact?**
-

**What are your project timelines?**
-

**Additional context**
-

If you have the same (or similar) feature request, please upvote this issue with thumbs up 👍
and use the comments section to provide answers to the questions above.

n9wxu, 2023-02-02:
sprintf and snprintf are both forbidden by MISRA rule 21.6.  The intent is for the kernel to be fully MISRA compliant without exceptions.  These print functions are used in the optional runtime statistics functions to fill a user supplied string buffer.

araujo88, 2023-08-05:
I believe that this [library](https://github.com/rurban/safeclib) provides a MISRA compliant alternative to `snprintf`. I guess this lib should be added on a separate PR unrelated to this issue specifically?

ydhuang28, 2023-08-08:
The only instances of `sprintf` are in `vTaskList` and `vTaskGetRunTimeStats` that are (albeit in `tasks.c`) not considered to be part of the FreeRTOS kernel since their purpose are utility functions meant to produce human-readable output; thus, they may not have to be MISRA-compliant (? @n9wxu). We could move ahead with a PR that replaces `sprintf` with `snprintf` and that would solve the immediate issue.

ydhuang28, 2023-08-08:
@Hadatko Are you using the `vTaskList` and `vTaskGetRunTimeStats` functions? If not, simply set `configUSE_STATS_FORMATTING_FUNCTIONS = 0` in your `FreeRTOSConfig.h`, and the complier should stop complaining.

Otherwise...

The parameter(s) of the API of `vTaskList` and `vTaskGetRunTimeStats` prevent us from using `snprintf` properly since they're only given a pointer to a `char` array of unknown length. A couple suggestions (with backwards-compatibility in mind, i.e., no API changes):

#### Approach 1

- We introduce a new config macro for the length of the task printing buffer (e.g. something like `#define configTASK_WRITE_BUFFER_LENGTH ( ( size_t ) 1024 )`)
- In `vTaskList` and `vTaskGetRunTimeStats`, do something like this:
```c
/* ... getting task status with uxTaskGetSystemState ... */
#ifdef configTASK_WRITE_BUFFER_LENGTH
{
sprintf( pcWriteBuffer, "\t%c\t%u\t%u\t%u\r\n", ... );
}
#else
{
snprintf( pcWriteBuffer, configTASK_WRITE_BUFFER_LENGTH, "\t%c\t%u\t%u\t%u\r\n", ... );
}
#endif
```

#### Approach 2

We manually write to the buffer without using any `stdio.h` functions and put the buffer writing code in a `prv` helper function. This would still require somehow knowing how long the buffer is, so a config macro might still be necessary.

@Hadatko if you are using `vTaskList` and/or `vTaskGetRunTimeStats` in its current state, let us know if the approaches looks good.

araujo88, 2023-08-17:
Added macro `configTASK_WRITE_BUFFER_LENGTH` on this [PR](https://github.com/FreeRTOS/FreeRTOS/pull/1071).

Hadatko, 2023-08-17:
HI guys. Sorry for not being answering. I am not directly using mentioned functions. But i found configUSE_STATS_FORMATTING_FUNCTIONS set to 1. Now i putted it to 0 (thank you for heads up). I posted it here as i am using this RTOS so i wanted report issue once noticed ;)

Thank you for your PR. It is great that you want solve this issue ;)

kar-rahul-aws, 2023-09-28:
Hi @Hadatko 
Closing this feature request, since the PR #802   which addresses this request has been merged.

closed 2023-09-28 by kar-rahul-aws