FreeRTOS/FreeRTOS-Kernel #622: [Feature Request] add FreeRTOS to conan center

JuliusCaesar89, 2023-02-10:
Hey guys,
since FreeRTOS has now official CMake Support it would be nice to add FreeRTOS to conan center.
https://conan.io/center/
Conan is a package manager for C/C++. It makes consuming 3rd parties very easy.
By adding FreeRTOS to conan center it will be much easier to be integrated in multiple projects.

greetins Julian

aggarg, 2023-02-12:
I have not used conan center before, so my assessment may be wrong. Conan center seems to be a place focusing on publishing binaries rather than source packages whereas FreeRTOS is consumed as source because it needs a configuration file (`FreeRTOSConfig.h`) from the application writer. Would you please elaborate how you suggest publishing FreeRTOS on conan center.

Thanks.

JuliusCaesar89, 2023-02-12:
Hey @aggarg thx for your feedback :)
Conan is capable of handling binaries and sources for many different build systems.
Conan Center does not host binaries, it just hosts the Conan Recipes which tell the conan clients how to build the source. (In the case of FreeRTOS using CMake).
From a single conan recipe FreeRTOS conan can build FreeRTOS for many different platforms via calling `conan install FreeRTOS`
https://docs.conan.io/en/2.0/tutorial/consuming_packages/build_simple_cmake_project.html
In your project you can just call conan install FreeRTOS or use the provided conan cmake wrapper to add FreeRTOS. This makes the integration of FreeRTOS in different projects very easy.
Conan is like Maven or Gradle in Java ;)

You are right the Config Header is a bit of a challenge here.
The easiest way should be providing a path to FreeRTOSConfig.h as a conan option and add the hash of FreeRTOSConfig.h to the conan options.

A very simple Conan Recipe looks like this
```

from conan import ConanFile
from conan.tools.cmake import CMakeToolchain, CMake, cmake_layout
from conan.tools.scm import Git

class FreeRTOSConan(ConanFile):
    name = "FreeRTOS"
    version = "10.5.1"

    # Optional metadata
    license = "Apache 2.0"
    author = "Julian Heni"
    url = "https://github.com/FreeRTOS/FreeRTOS-Kernel"
    description = "FreeRTOS as a conan 3rd party"
    topics = ("FreeRTOS", "c", "c++")

    # Binary configuration
    settings = "os", "compiler", "build_type", "arch"
    options = {"shared": [True, False], "fPIC": [True, False], "BUILD_TARGET": [True, False]}
    default_options = {"shared": True, "fPIC": True, "BUILD_TARGET": False}
    generators = "CMakeDeps"
    # build_policy = "always"  # for better debugging always build freertos

    def source(self):
        git = Git(self)
        # check out tagged version
        clone_args = [f"-b V{FreeRTOSConan.version}"]
        git.clone(url="https://github.com/FreeRTOS/FreeRTOS-Kernel.git", args=clone_args)

    def config_options(self):
        if self.settings.os == "Windows":
            del self.options.fPIC

    def layout(self):
        cmake_layout(self)

    def generate(self):
        tc = CMakeToolchain(self)
        if self.options.BUILD_TARGET:
            print("Building target")
            tc.cache_variables["BUILD_TARGET"] = "ON"

        else:
            print("Building simulation")
            tc.cache_variables["BUILD_TARGET"] = "OFF"
        tc.generate()

    def build(self):
        cmake = CMake(self)
        cmake.configure(build_script_folder="freertos")
        cmake.build()
        cmake.test(build_type=self.settings.build_type, target="FreeRTOS")

    def package(self):
        cmake = CMake(self)
        cmake.install()

    def package_info(self):
        self.cpp_info.libs = ["FreeRTOS"]
        self.cpp_info.includedirs = ["include"]

```

As you can see we are just invoking CMake. Since FreeRTOS has CMake Support nowit should be easy to add it.

aggarg, 2023-02-14:
Thank you for sharing the details. We currently do not have any plans of publishing to conan center but we will keep you posted if that changes in future. 

In the meanwhile, if you want to contribute and publish yourself, is there anything that you need from us?

JuliusCaesar89, 2023-02-18:
Hey @aggarg 
thx for your feedback. Currently, I do not need anything from you guys ;) However if I succeed a little promotion on the FreeRTOS website would be nice ;)

jputcu, 2023-08-25:
@JuliusCaesar89 Somebody is making an attempt: https://github.com/conan-io/conan-center-index/pull/19399

JuliusCaesar89, 2023-08-26:
ah thx for the info :)