FreeRTOS/FreeRTOS-Kernel #628: [BUG] FreeRTOS/FreeRTOS/Demo/CORTEX_MPU_M3_MPS2_QEMU_GCC demo build failure

urutva, 2023-02-20:
**Describe the bug**
Building the full demo FreeRTOS/FreeRTOS/Demo/CORTEX_MPU_M3_MPS2_QEMU_GCC with `make FULL_DEMO=1` with latest [kernel](https://github.com/FreeRTOS/FreeRTOS-Kernel/commit/8252edec4547afed4e6ada74bd9353b52972894f) fails with following compilation error:
```
/tmp/tmp-test/FreeRTOS/FreeRTOS/Source/event_groups.c: In function 'vEventGroupDelete':
/tmp/tmp-test/FreeRTOS/FreeRTOS/Source/event_groups.c:59:46: error: conversion from 'long unsigned int' to 'TickType_t' {aka 'const short unsigned int'} changes value from '33554432' to '0' [-Werror=overflow]
   59 |     #define eventUNBLOCKED_DUE_TO_BIT_SET    0x02000000UL
      |                                              ^~~~~~~~~~~~
/tmp/tmp-test/FreeRTOS/FreeRTOS/Source/event_groups.c:652:88: note: in expansion of macro 'eventUNBLOCKED_DUE_TO_BIT_SET'
  652 |             vTaskRemoveFromUnorderedEventList( pxTasksWaitingForBits->xListEnd.pxNext, eventUNBLOCKED_DUE_TO_BIT_SET );
```

**Target**
- Development board: M3 MPS2 QEMU
- Instruction Set Architecture: Armv7-M
- IDE and version: None
- Toolchain and version: arm-none-eabi-gcc (Arm GNU Toolchain 12.2.Rel1 (Build arm-12.24)) 12.2.1 20221205

**Host**
- Host OS: Ubuntu
- Version: 20.04

**To Reproduce**
- git clone git@github.com:FreeRTOS/FreeRTOS.git --recurse-submodules
- cd FreeRTOS/FreeRTOS/Source
- git fetch -all
- git checkout main
- cd ../Demo/CORTEX_MPU_M3_MPS2_QEMU_GCC
- make FULL_DEMO=1 

**Expected behavior**
Build succeeds without failure

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Digging further into this, the build fails because the `TickType_t` gets `typedef`ed to uint16_t instead of uint32_t at https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/portable/GCC/ARM_CM3_MPU/portmacro.h#L60 even though [configUSE_16_BIT_TICKS is set to 0](https://github.com/FreeRTOS/FreeRTOS/blob/main/FreeRTOS/Demo/CORTEX_MPU_M3_MPS2_QEMU_GCC/FreeRTOSConfig.h#L60). 

It seems like the backward compatibility support for `configUSE_16_BIT_TICKS` added as part of adding support for 64-bit events https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/597 isn't working as expected. The PR introduces a new macro `configTICK_TYPE_WIDTH_IN_BITS` which is undefined when `portmacro.h` is included in [FreeRTOS.h](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/include/FreeRTOS.h#L70). The macro gets defined further down in [FreeRTOS.h](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/include/FreeRTOS.h#L171) to support backward compatibility for `configUSE_16_BIT_TICKS` macro. Since the macro `configTICK_TYPE_WIDTH_IN_BITS` isn't undefined when `portmacro.h` is included, `configTICK_TYPE_WIDTH_IN_BITS` is evaluated as `0` in the [expression](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/portable/GCC/ARM_CM3_MPU/portmacro.h#L60) which causes the equality to succeed and `TickType_t` to typedef'ed to `uint16_t`. The undefined macros can be checked by enabling `-Wundef` after https://github.com/FreeRTOS/FreeRTOS/blob/main/FreeRTOS/Demo/CORTEX_MPU_M3_MPS2_QEMU_GCC/Makefile#L45.

urutva, 2023-02-20:
One of the possible solutions is to move the check and definition https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/include/FreeRTOS.h#L163 - https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/include/FreeRTOS.h#L185 before inclusion of `portable.h` https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/include/FreeRTOS.h#L70.

amazonKamath, 2023-02-23:
@urutva  @aggarg I guess we can close this issue now?

aggarg, 2023-02-23:
Yes. Closing the issue.

closed 2023-02-23 by aggarg