FreeRTOS/FreeRTOS-Kernel #632: [BUG] commit 8cd5451 broken build

RodrigoDornelles, 2023-02-23:
**Describe the bug**
I'm using freertos kernel in a project, and I noticed that at the last minute the unit tests started to fail, well I referenced cmake to download the commit `5d05601` and it solved the problem.

 - :x: 8cd5451 not compile.
 - ✔️  5d05601 compile. 

**Target**
- Development board: Posix (Desktop)
- Instruction Set Architecture: x86_64 (Desktop)
- IDE and version: gcc (Alpine 11.2.1_git20220219) 11.2.1 20220219

**Host**
- Host OS: Linux 2d309816c9b4 5.15.79.1-microsoft-standard-WSL2
- Version: Alpine 3.16.2

**To Reproduce**

### cmake config ###
```cmake
include_directories("${CMAKE_SOURCE_DIR}/vendor/freertos-kernel/include")
include_directories("${CMAKE_SOURCE_DIR}/vendor/freertos-kernel/portable/ThirdParty/GCC/Posix")
include_directories("${CMAKE_SOURCE_DIR}/vendor/freertos-kernel/portable/ThirdParty/GCC/Posix/utils")

set(FREERTOS_HEAP "4")
set( FREERTOS_PORT "GCC_POSIX" CACHE STRING "" FORCE)

add_library(freertos_config INTERFACE)

target_include_directories(freertos_config SYSTEM
INTERFACE
    tests
)

target_compile_definitions(freertos_config
  INTERFACE
    projCOVERAGE_TEST=0
)

# Fetch FreeRTOS
FetchContent_Declare( freertos_kernel
  GIT_REPOSITORY "https://github.com/FreeRTOS/FreeRTOS-Kernel"
  GIT_TAG        "origin/main"
  SOURCE_DIR    "${CMAKE_SOURCE_DIR}/vendor/freertos-kernel"
)

FetchContent_MakeAvailable(freertos_kernel)
```

### commands ### 

```
cmake .
make
```

**Expected behavior**
Succeful compile.

**Screenshots**

```
Consolidate compiler generated dependencies of target freertos_kernel_port
[ 11%] Building C object _deps/freertos_kernel-build/portable/CMakeFiles/freertos_kernel_port.dir/ThirdParty/GCC/Posix/port.c.o
In file included from /app/vendor/freertos-kernel/portable/ThirdParty/GCC/Posix/port.c:65:
/app/vendor/freertos-kernel/include/FreeRTOS.h:282:6: error: #warning is a GCC extension [-Werror]
  282 |     #warning Co-routines have been removed from FreeRTOS-Kernel versions released after V10.5.1. You can view previous versions of the FreeRTOS Kernel at github.com/freertos/freertos-kernel/tree/V10.5.1 .
      |      ^~~~~~~
/app/vendor/freertos-kernel/include/FreeRTOS.h:282:6: error: #warning Co-routines have been removed from FreeRTOS-Kernel versions released after V10.5.1. You can view previous versions of the FreeRTOS Kernel at github.com/freertos/freertos-kernel/tree/V10.5.1 . [-Werror=cpp]
cc1: all warnings being treated as errors
make[2]: *** [_deps/freertos_kernel-build/portable/CMakeFiles/freertos_kernel_port.dir/build.make:76: _deps/freertos_kernel-build/portable/CMakeFiles/freertos_kernel_port.dir/ThirdParty/GCC/Posix/port.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1070: _deps/freertos_kernel-build/portable/CMakeFiles/freertos_kernel_port.dir/all] Error 2
make: *** [Makefile:101: all] Error 2
```

RodrigoDornelles, 2023-02-23:
I removed the define from `configUSE_CO_ROUTINES` and managed to compile, but I'll leave the issue open for the maintainers to be aware of and decide if it's worth turning warnings into errors.

dachalco, 2023-02-24:
Thanks @RodrigoDornelles for reporting.

I think `-Werror` alone may have broken some existing builds. Let me review with the team on next steps. Inclusion of `-Werror` may have been intentional, to hard stop introduction of further warnings which previously may have been overlooked. Turning off the faucet, so we can drain the sink -- so to speak.

aggarg, 2023-02-27:
This warning is coming from this line - https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/include/FreeRTOS.h#L281

It is to tell the user that co-routines are no longer present and therefore, the application must not use them. If you do not use co-routines, removing the `configUSE_CO_ROUTINES` from your `FreeRTOSConfig.h` is the right fix.

closed 2023-02-28 by dachalco