FreeRTOS/FreeRTOS-Kernel #646: [BUG] Build With CMake and GCC as static library

0x7a657573, 2023-03-18:
**Describe the bug**
I tried to add FreeRTOS on my CMake Project but in compile time I get some error like this:

[build] ../system/FreeRTOS/portable/GCC/ARM_CM3/portmacro.h:135:51: error: unknown type name 'inline'
[build]          __attribute__( ( always_inline ) ) static inline uint8_t ucPortCountLeadingZeros( uint32_t ulBitmap )

I think this error happen because in the FreeRTOS, CMakelists.txt at line 296 say to compiler use C90 standard.
`set_property(TARGET freertos_kernel PROPERTY C_STANDARD 90)
`
when I change this line to use GNU99 standard, error is gone !

**Target**
- Development board: [STM32F103CBT6]
- Instruction Set Architecture: [e.g. RV32IMAC]
- IDE and version: [vscode]
- Toolchain and version: [GCC arm-none-eabi-8.3.1]

**Host**
- Host OS: [Linux]
- Version: [Debian 11.6]

RichardBarry, 2023-03-18:
Grateful if you can create a support thread on this at https://forums.freertos.org until we determine the route cause as the kernel should build with C90, and GCC is happy with "inline".  Thanks.

0x7a657573, 2023-03-19:
I posted my Issue to [forums.freertos.org](https://forums.freertos.org/t/freertos-gcc-cmake/16984)

closed 2023-03-28 by aggarg