FreeRTOS/FreeRTOS-Kernel #647: [Feature Request] add custom hook on tick overflow/correction

Viatorus, 2023-03-20:
**Is your feature request related to a problem? Please describe.**

We are tracking the _uptime_ of the system with the SysTick of the OS.

Since the tick count of freeRTOS does overflow from time to time (~49 days for a 32 bit timer), we are using using the `vApplicationTickHook` to increment a 64 bit value by ourself.

This has some drawbacks:
* Overhead for for very similar tasks: incrementing a tick
* doesn't work with tickless idle modes -> requires manual adjusting of the _uptime_ after a tickless time > 1 ms.

**Describe the solution you'd like**

I think one solution would be to get notified (through a configurable hook) whenever the OS tick overflows (on tick increment) or is adjusted (after an sleep in the idle task).

**How many devices will this feature impact?**
Configurable through a define, option so would be no impact if not used.

If you have the same (or similar) feature request, please upvote this issue with thumbs up 👍
and use the comments section to provide answers to the questions above.

aggarg, 2023-06-20:
We recently introduced a new config option `configTICK_TYPE_WIDTH_IN_BITS ` which can be defined to `TICK_TYPE_WIDTH_64_BITS` to change the type of `TickType_t` to a 64-bit type. Would that be helpful for your usecase?

aggarg, 2023-07-03:
@Viatorus Would you please take a look at the option I suggested above?

Viatorus, 2023-07-04:
@aggarg That sounds like something, we can definitely could use. :) Thank you!

closed 2023-07-04 by Viatorus