FreeRTOS/FreeRTOS-Kernel #661: [BUG]  Where is croutine.c in the main branch?

jwestmoreland, 2023-04-14:
Hello All,

Looked in the docs - at least what I could find in the rev history - has croutine.c been removed?

Just wondering - maybe I missed something.

It's there in the v10.5.1 branch/tag.

Thanks!
John W.

rawalexe, 2023-04-14:
Hello John W.,
We have depreciated croutine.c you can find the place holder [here](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/include/FreeRTOS.h#:~:text=%23warning%20Co%2Droutines%20have%20been%20removed%20from%20FreeRTOS%2DKernel%20versions%20released%20after%20V10.5.1.%20You%20can%20view%20previous%20versions%20of%20the%20FreeRTOS%20Kernel%20at%20github.com/freertos/freertos%2Dkernel/tree/V10.5.1%20). Please use v10.5.1 if you still need them.

Please let me know if have any other queries.

Best Regards,
Anubhav Rawal

jwestmoreland, 2023-04-14:
Anubhav,

Has the functionality in croutine.c been incorporated someplace else?

Thanks,
John W.

rawalexe, 2023-04-14:
Greetings John, 
We noticed that the c file hadn't been touched in a long time and we never had a questions or pull requests on it. After a careful considerations we felt that it was safe to remove the file as most folks were using larger processors and weren't in need of the file. However if you still need them, you can find them from the [v10.5.1 release ](https://github.com/FreeRTOS/FreeRTOS-Kernel/releases/tag/V10.5.1). As the file has been stable for a long time you can still use the past version of the file.

Hopefully this helps. Please let us know if you have more questions.

Best Regards,
Anubhav Rawal

jwestmoreland, 2023-04-15:
Anubhav,

OK - so, no one's using a stack challenged processor these days, OK.

Or - some others may notice when 10.5.1+ is released.

Hopefully Richard sees this comment and chimes in.

Thanks,
John W.

RichardBarry, 2023-04-15:
Hi John - noted - I will discuss with the team Monday.

jwestmoreland, 2023-04-15:
Hello Richard,

Interested to hear what's decided.

I have a few things in the pipeline here - on one especially that's 'ram challenged' - pretty sure we will end up using co-routines.

Thanks!
John

RichardBarry, 2023-04-17:
Hi John - we decided to add the file back into the git repo, and then update co-routine related website pages noting co-routines are no longer under development.  I hope that is agreeable for you.

jwestmoreland, 2023-04-17:
Hello Richard,

That's fine.

Thanks!
John

AniruddhaKanhere, 2023-06-02:
Since the above PR has been merged, we will close this issue. Please feel free to reopen this thread if you feel that your issue wasn't resolved correctly.

closed 2023-06-02 by AniruddhaKanhere