FreeRTOS/FreeRTOS-Kernel #687: [BUG] circular dependency in CMake causes linker errors

CrustyAuklet, 2023-06-10:
**Describe the bug**
There is a circular dependency between CMake targets `freertos_kernel` and `freertos_kernel_port`, causing the final executable to have both archive files linked twice. This causes multiple definition errors if the consuming program uses `--whole-archive`

**Target**
Custom SAML21 board (ARM Cortex-M0) and UNIX port simulator.
Toolchain: gcc-12

**Host**
- Host OS: Ubuntu 20.04, Windows11
- Version: [e.g. Mojave 10.14.6]

**To Reproduce**
- Create a cmake static library that links against `freertos_kernel`
- Create an executable target that links with the static library using `--whole-archive`
- observe multiple definition errors, and in the linker command `freertos_kernel.a` and `freertos_kernel_port.a` are emitted twice each.

**Expected behavior**
each static library should only be in the final link command once

**Additional context**
This is important if creating a "framework" or "SDK" using FreeRTOS. in my static library is the `main()` and `vApplicationGetIdleTaskMemory` functions, reset vectors, etc. Some of these functions will not be linked into the final binary without using `--whole-archive`, but using that will cause errors due to the circular reference in FreeRTOS.

I solved this in my framework, which also has a backend/port structure, by splitting the main target into 2. One interface only library that has the include paths and defines, and the actual main target with all the source files that links against that interface library. `freertos_kernel_port` could now link against that interface library and get the include paths, without depending on the actual `freertos_kernel` static library.

cookpate, 2023-06-10:
Hello, our team will investigate your use case.

closed 2023-07-10 by cookpate

CrustyAuklet, 2023-07-11:
cool! :)