FreeRTOS/FreeRTOS-Kernel #697: [BUG] Missing portPOINTER_SIZE_TYPE  definition for ATmega port

jputcu, 2023-06-28:
The Atmega port: `portable/ThirdParty/GCC/ATmega/portmacro.h` needs the following:
~~~~
#define portPOINTER_SIZE_TYPE uint16_t
~~~~

kstribrnAmzn, 2023-06-28:
Thank you for raising this issue! While I'm happy to make the change on your behalf, would you like to make the change? In doing so you'll get to contribute a fix for a bug you've identified.

kstribrnAmzn, 2023-06-28:
The ports _should be extremely similar_. The key difference is how these ports got created. The ATMega323 port was most likely created by the FreeRTOS team. The more general, third party ATMega port was actually created in [this PR](https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/48) by a contributing community member.

As mentioned in that PR, there are several ATMega chips which aren't supported. Digging through the commits to the port files, it looks like these chips are still not supported as no further work to them has been done.

If additional ports are needed, we'd love for you to either submit a PR contributing them or create a feature request for whatever ports you need. We use feature requests to understand user needs.

PR #698 (2023-06-29): Fix #697: Missing portPOINTER_SIZE_TYPE definition for ATmega port
Related Issue
-----------
Fix #697: Missing portPOINTER_SIZE_TYPE definition for ATmega port

By submitting this pull request, I confirm that you can use, modify, copy, and redistribute this contribution, under the terms of your choice.
commit 1c5eca34 (2023-06-29): Fix #697: Missing portPOINTER_SIZE_TYPE definition for ATmega port (#698)
closed 2023-06-29 by kstribrnAmzn