FreeRTOS/FreeRTOS-Kernel #710: [BUG] Windows 64-bit MingW error in xQueueGenericCreate port due to comparison of different types

jputcu, 2023-07-19:
Using the current `master` and the `FreeRTOS\portable\MSVC-MingW` port on Windows 64-bit machine I get the following error:
~~~~
FreeRTOS/queue.c: In function 'xQueueGenericCreate':
FreeRTOS/queue.c:489:48: error: comparison is always true due to limited range of data type [-Werror=type-limits]
  489 |             ( ( SIZE_MAX - sizeof( Queue_t ) ) >= ( uxQueueLength * uxItemSize ) ) )
      |                                                ^~
cc1.exe: all warnings being treated as errors
~~~~

The right hand side: `UBaseType_t` is an `unsigned long`(32-bit). The left side is `uint64_t`(64-bit). Sizes don't match.

aggarg, 2023-07-21:
Are you choosing `x64` as the platform in your VS project? Can you try changing it to WIN32?

[image]

jputcu, 2023-07-21:
I'm using `msys2/ucrt64` on the command line with `gcc`. I think we need to cast the left side to `UBaseType_t` or make that type a `size_t` or `unsigned long long` in the port, but I don't know what that would impact.

aggarg, 2023-07-21:
Can you try changing [these lines](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/portable/MSVC-MingW/portmacro.h#L49-L50) to the following - 

```
#if defined( __x86_64__) || defined( _M_X64 )
    typedef long long BaseType_t;
    typedef unsigned long long UBaseType_t;
#else
    typedef long BaseType_t;
    typedef unsigned long UBaseType_t;
#endif
```

jputcu, 2023-07-21:
I currently don't have access to my work (Windows) PC, but I feels we are actually just recreating `size_t`.

RichardBarry, 2023-07-21:
Sorry - not in a position to look at the actual code or reproduce the error right now, but concur other places in the code use size_t to switch between 32 and 64-bit architectures.

aggarg, 2023-07-24:
@jputcu Were you able to try out my suggestion?

jputcu, 2023-07-24:
@aggarg I just tried it and the compiler warning is gone.

aggarg, 2023-07-24:
Thanks for confirming. I have created the following PR to fix this - https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/715

closed 2023-07-24 by aggarg