FreeRTOS/FreeRTOS-Kernel #714: [BUG] pxPortInitialiseStack OverWrites First Word of TLS Block

bebebib, 2023-07-21:
**Describe the bug**
With Thread Local Storage (TLS) enabled and stack growth is less than 0, when [configINIT_TLS_BLOCK ](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/include/picolibc-freertos.h#L65)is called from [prvInitialiseNewTask](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/tasks.c#L961), it initializes a TLS block at the top of the task stack. Then, the function pxPortInitialiseStack (in some ports, such as the [ARM_CR5](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/portable/GCC/ARM_CR5/port.c#L288)), sets the top of the stack to a certain value (none or other).

This effectively overwrites the first word of the TLS block, since the top of the stack and the first word of the TLS block overlap. In all ports of FreeRTOS, the function pxPortInitialiseStack should decrement the stack first before writing any values, as otherwise it will overwrite the first value of the TLS block

**Target**
- Development board: Cortex-R5 processor (and others)
- Instruction Set Architecture: ARM
- IDE and version: VS Code
- Toolchain and version: gcc-arm-none-eabi 11.2.0

**Host**
- Host OS: Windows / WSL

**To Reproduce**
- Configure FreeRTOS with configUSE_C_RUNTIME_TLS_SUPPORT and `portSTACK_GROWTH < 0`, and attempt to create a global TLS variable of at least one word in size, initialize it to a value other than 0 (in this example 5)

**Expected behavior**
When we create a task and start the scheduler, said task should read the global TLS variable as the value it was initialized at (in this case, 5).

**Actual behavior**
The global TLS value is read as 0 in every task that uses it (makes a thread local copy). 

**Example**
This is a simple order of operations for what happens in configINIT_TLS_BLOCK when `portSTACK_GROWTH < 0`
[image]

aggarg, 2023-07-26:
You are right that the `pxPortInitialiseStack` should not write to the `pxTopOfStack` byte. The following is from the ARM manual - 
```
The stack pointer holds the address of the last stacked item in memory. When
the processor pushes a new item onto the stack, it decrements the stack pointer
and then writes the item to the new memory location.
```
Therefore, the right thing to do, as you suggested, is to first decrement `pxTopOfStack` and then write. I checked that we are 
doing so in many ports and changing all those ports would be a massive change. Another workaround can be to ensure that `xTLSBlock` and `pxTopOfStack` do not overlap in `configINIT_TLS_BLOCK`. Would you please try changing [these lines](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/include/picolibc-freertos.h#L65-L73) with the following code block and let me know if that fixes the problem - 

```c
    #define configINIT_TLS_BLOCK( xTLSBlock, pxTopOfStack )                                     \
    do {                                                                                        \
        xTLSBlock = ( void * ) ( ( ( ( portPOINTER_SIZE_TYPE ) pxTopOfStack ) -                 \
                                   picolibcTLS_SIZE ) &                                         \
                                 ~picolibcTLS_ALIGNMENT_MASK );                                 \
        pxTopOfStack = ( StackType_t * ) ( ( ( ( portPOINTER_SIZE_TYPE ) xTLSBlock ) - 1 ) &    \
                                           ~picolibcSTACK_ALIGNMENT_MASK );                     \
        _init_tls( xTLSBlock );                                                                 \
    } while( 0 )
```

bebebib, 2023-07-27:
I have tried something similar already with the `configINIT_TLS_BLOCK` macro that did fix this issue:
```
 do {                                                                                               \
        pxTopOfStack = ( StackType_t * ) ( ( ( portPOINTER_SIZE_TYPE ) pxTopOfStack ) - _tls_size());  \
        /* Align down to the nearest multiple of 8 */                                                  \
        pxTopOfStack = ( StackType_t * ) ( ( ( portPOINTER_SIZE_TYPE ) pxTopOfStack ) & ( ~( ( portPOINTER_SIZE_TYPE ) portBYTE_ALIGNMENT_MASK ) ) ); \
        xTLSBlock = pxTopOfStack;                                                                      \
        _init_tls( xTLSBlock );                                                                        \
        /* Move stack one word down so that pxTopOfStack and xTLSBlock do not overlap */               \
        pxTopOfStack = ( StackType_t * ) ( ( ( portPOINTER_SIZE_TYPE ) pxTopOfStack ) - portBYTE_ALIGNMENT);  \
    } while( 0 )
```
Should a formal PR be submitted for this fix? This change should not effect the other ports that decrement the stack pointer already, correct? They would just end up decrementing the stack pointer twice.

aggarg, 2023-07-28:
> This change should not effect the other ports that decrement the stack pointer already, correct? They would just end up decrementing the stack pointer twice.

Yes, that is correct. 

I'd suggest to use the code block I suggested mostly for readability reasons - 
1. xTLSBlock  needs to be aligned on picolibcTLS_ALIGNMENT_MASK.
2. pxTopOfStack  needs to be aligned on picolibcSTACK_ALIGNMENT_MASK. 

Would you please test the change I suggested and raise a PR?

aggarg, 2023-08-03:
@bebebib Do you want to raise the PR or do you want us to do?

bebebib, 2023-08-03:
I will raise one, but I will do it with a different account if that is alright

aggarg, 2023-08-03:
Sure, that works.

chinglee-iot, 2023-08-07:
Close this issue due the PR is merged.

closed 2023-08-07 by chinglee-iot