FreeRTOS/FreeRTOS-Kernel #718: [BUG] Conversion errors on atmega port since `-Wconversion` commit

jputcu, 2023-07-24:
Using the latest FreeRTOS compilation starts failing due to the added `-Wconversion`.

~~~~
FreeRTOS/queue.c: In function 'xQueueGiveFromISR':
FreeRTOS\queue.c(1332,42): error: conversion to 'UBaseType_t {aka volatile unsigned char}' from 'int' may alter its value [-Werror=conversion]
		             pxQueue->uxMessagesWaiting = uxMessagesWaiting + ( UBaseType_t ) 1;
		                                          ^~~~~~~~~~~~~~~~~
		FreeRTOS/queue.c: In function 'xQueueReceive':
FreeRTOS\queue.c(1477,46): error: conversion to 'UBaseType_t {aka volatile unsigned char}' from 'int' may alter its value [-Werror=conversion]
		                 pxQueue->uxMessagesWaiting = uxMessagesWaiting - ( UBaseType_t ) 1;
		                                              ^~~~~~~~~~~~~~~~~
		FreeRTOS/queue.c: In function 'xQueueSemaphoreTake':
FreeRTOS\queue.c(1634,46): error: conversion to 'UBaseType_t {aka volatile unsigned char}' from 'int' may alter its value [-Werror=conversion]
		                 pxQueue->uxMessagesWaiting = uxSemaphoreCount - ( UBaseType_t ) 1;
		                                              ^~~~~~~~~~~~~~~~
		FreeRTOS/queue.c: In function 'xQueueReceiveFromISR':
FreeRTOS\queue.c(2006,42): error: conversion to 'UBaseType_t {aka volatile unsigned char}' from 'int' may alter its value [-Werror=conversion]
		             pxQueue->uxMessagesWaiting = uxMessagesWaiting - ( UBaseType_t ) 1;
		                                          ^~~~~~~~~~~~~~~~~
		FreeRTOS/queue.c: In function 'uxQueueSpacesAvailable':
FreeRTOS\queue.c(2140,20): error: conversion to 'UBaseType_t {aka unsigned char}' from 'int' may alter its value [-Werror=conversion]
		         uxReturn = pxQueue->uxLength - pxQueue->uxMessagesWaiting;
		                    ^~~~~~~
		FreeRTOS/queue.c: In function 'prvGetDisinheritPriorityAfterTimeout':
FreeRTOS\queue.c(2253,47): error: conversion to 'UBaseType_t {aka unsigned char}' from 'int' may alter its value [-Werror=conversion]
		             uxHighestPriorityOfWaitingTasks = ( UBaseType_t ) configMAX_PRIORITIES - ( UBaseType_t ) listGET_ITEM_VALUE_OF_HEAD_ENTRY( &( pxQueue->xTasksWaitingToReceive ) );
		                                               ^
		FreeRTOS/queue.c: In function 'prvCopyDataToQueue':
FreeRTOS\queue.c(2343,34): error: conversion to 'UBaseType_t {aka volatile unsigned char}' from 'int' may alter its value [-Werror=conversion]
		     pxQueue->uxMessagesWaiting = uxMessagesWaiting + ( UBaseType_t ) 1;
		                                  ^~~~~~~~~~~~~~~~~
		cc1.exe: all warnings being treated as errors
~~~~

jputcu, 2023-07-24:
The casts to `UBaseType_t` should be done on the mathematics results. Casting the parts doesn't make sense because the compiler will cast them back to e.g. `unsigned int` for the math.

kar-rahul-aws, 2023-07-25:
Hi @jputcu 
Thanks for reporting the issue.  We are working towards a fix and will keep you updated.

kar-rahul-aws, 2023-07-26:
Hi @jputcu 
A [patch](https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/720) has been applied for fixing the changes. 
Please let us know if this fix works well for you.

kar-rahul-aws, 2023-07-26:
Thanks for the update @jputcu . 
Closing this issue for now, please let us know if have any more queries.

closed 2023-07-26 by kar-rahul-aws