FreeRTOS/FreeRTOS-Kernel #731: [BUG]

pranjalchanda08, 2023-07-30:
**Describe the bug**

I am currently seeing this issue in the main Branch of FreeRTOS Kernel:

[109/111] Building C object CMakeFiles/JadeDrive.dir/3rdParty/FreeRTOS-Kernel/queue.c.obj
In file included from ../3rdParty/FreeRTOS-Kernel/include/FreeRTOS.h:64,
                 from ../3rdParty/FreeRTOS-Kernel/queue.c:37:
../3rdParty/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/include/portmacro.h: In function 'vPortRecursiveLock':
../3rdParty/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/include/portmacro.h:197:32: warning: comparison of unsigned expression >= 0 is always true [-Wtype-limits]
  197 |         configASSERT(ulLockNum >= 0 && ulLockNum < portRTOS_SPINLOCK_COUNT );

_________________________________________________________________________________________________________________
Since, my workspace treats all Warnings as errors this small issue is creating the builds issue.
This issue is seen on main and smp branch of FreeRTOS-Kernel

**Target**
RPi Pico

**Host**
- Host OS: Ubuntu
- Compiler: GCC: 9.2.1

**To Reproduce**
Build the FreeRTOS Kernel for RPI Pico with -Wall -Wextra -Weverything flags on

**Expected behavior**
The warning shall not be produced. A clean build

ActoryOu, 2023-07-31:
Hi @pranjalchanda08,
Thanks for bringing this into our attention.

I've created a PR #732 to fix this compile warning.
Will close this issue when it gets merged.

Thanks.

pranjalchanda08, 2023-07-31:
@ActoryOu,

Its coming up in task.c

Even found this issue if can be rectified in the same PR:
```sh
../3rdParty/FreeRTOS-Kernel/tasks.c: In function 'prvYieldCore':
../3rdParty/FreeRTOS-Kernel/tasks.c:763:62: warning: comparison of integer expressions of different signedness: 'BaseType_t' {aka 'long int'} and 'uint' {aka 'unsigned int'} [-Wsign-compare]
  763 |         if( ( portCHECK_IF_IN_ISR() == pdTRUE ) && ( xCoreID == portGET_CORE_ID() ) )
      |                                                              ^~
../3rdParty/FreeRTOS-Kernel/tasks.c:771:29: warning: comparison of integer expressions of different signedness: 'BaseType_t' {aka 'long int'} and 'uint' {aka 'unsigned int'} [-Wsign-compare]
  771 |                 if( xCoreID == portGET_CORE_ID() )
      |                             ^~
../3rdParty/FreeRTOS-Kernel/tasks.c: In function 'vTaskSuspend':
../3rdParty/FreeRTOS-Kernel/tasks.c:2832:44: warning: comparison of integer expressions of different signedness: 'TaskRunning_t' {aka 'long int'} and 'uint' {aka 'unsigned int'} [-Wsign-compare]
 2832 |                     if( xTaskRunningOnCore == portGET_CORE_ID() )
      |  
```
[image]

ActoryOu, 2023-07-31:
Hi @pranjalchanda08,
Thanks for your feedback.

I've updated the PR with the fix on tasks.c.
Could you help to try it to check if any other warnings?

Thanks.

pranjalchanda08, 2023-07-31:
Just out of curiosity why the CI does not catch such compilation issues? Maybe treating all warnings as errors can catch most of such issues.
Or may be you need some multiple compilation tests for different FreeRTOS configs. There can surely be other configs that I am not enabling may catch later as a possibility.

I have not seen the CI though if its being done but some config might be ignored.

pranjalchanda08, 2023-07-31:
Just hopped into the CI pipeline output for RPI-pico and found no compilation commands are enabled:

```sh

2023-07-31T06:38:04.3542627Z [15/457] /usr/bin/arm-none-eabi-gcc -DFREE_RTOS_KERNEL_SMP=1 -DLIB_FREERTOS_KERNEL=1 -DLIB_PICO_BIT_OPS=1 -DLIB_PICO_BIT_OPS_PICO=1 -DLIB_PICO_DIVIDER=1 -DLIB_PICO_DIVIDER_HARDWARE=1 -DLIB_PICO_DOUBLE=1 -DLIB_PICO_DOUBLE_PICO=1 -DLIB_PICO_FLOAT=1 -DLIB_PICO_FLOAT_PICO=1 -DLIB_PICO_INT64_OPS=1 -DLIB_PICO_INT64_OPS_PICO=1 -DLIB_PICO_MALLOC=1 -DLIB_PICO_MEM_OPS=1 -DLIB_PICO_MEM_OPS_PICO=1 -DLIB_PICO_MULTICORE=1 -DLIB_PICO_PLATFORM=1 -DLIB_PICO_PRINTF=1 -DLIB_PICO_PRINTF_PICO=1 -DLIB_PICO_RUNTIME=1 -DLIB_PICO_STANDARD_LINK=1 -DLIB_PICO_STDIO=1 -DLIB_PICO_STDIO_UART=1 -DLIB_PICO_STDLIB=1 -DLIB_PICO_SYNC=1 -DLIB_PICO_SYNC_CRITICAL_SECTION=1 -DLIB_PICO_SYNC_MUTEX=1 -DLIB_PICO_SYNC_SEM=1 -DLIB_PICO_TIME=1 -DLIB_PICO_UTIL=1 -DPICO_BOARD=\"pico\" -DPICO_BUILD=1 -DPICO_CMAKE_BUILD_TYPE=\"Release\" -DPICO_CONFIG_RTOS_ADAPTER_HEADER=/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Source/portable/ThirdParty/GCC/RP2040/include/freertos_sdk_config.h -DPICO_COPY_TO_RAM=0 -DPICO_CXX_ENABLE_EXCEPTIONS=0 -DPICO_NO_FLASH=0 -DPICO_NO_HARDWARE=0 -DPICO_ON_DEVICE=1 -DPICO_TARGET_NAME=\"on_core_zero\" -DPICO_USE_BLOCKED_RAM=0 -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/OnEitherCore -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Source/portable/ThirdParty/GCC/RP2040/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Source/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/common/pico_base/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/build/generated/pico_base -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/boards/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/rp2_common/pico_platform/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/rp2040/hardware_regs/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/rp2_common/hardware_base/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/rp2_common/hardware_clocks/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/rp2040/hardware_structs/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/rp2_common/hardware_claim/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/rp2_common/hardware_sync/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/rp2_common/hardware_gpio/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/rp2_common/hardware_irq/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/common/pico_sync/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/common/pico_time/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/rp2_common/hardware_timer/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/common/pico_util/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/rp2_common/hardware_resets/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/rp2_common/hardware_pll/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/rp2_common/hardware_vreg/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/rp2_common/hardware_watchdog/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/rp2_common/hardware_xosc/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/rp2_common/hardware_exception/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/rp2_common/pico_multicore/include -I/home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/co [... cut ...] Party/Community-Supported-Demos/CORTEX_M0+_RP2040/pico-sdk/src/rp2_common/boot_stage2/include -mcpu=cortex-m0plus -mthumb -O3 -DNDEBUG -std=gnu11 -ffunction-sections -fdata-sections -MD -MT OnEitherCore/CMakeFiles/on_core_zero.dir/main.c.obj -MF OnEitherCore/CMakeFiles/on_core_zero.dir/main.c.obj.d -o OnEitherCore/CMakeFiles/on_core_zero.dir/main.c.obj -c /home/runner/work/FreeRTOS-Kernel/FreeRTOS-Kernel/FreeRTOS/Demo/ThirdParty/Community-Supported-Demos/CORTEX_M0+_RP2040/OnEitherCore/main.c
```

pranjalchanda08, 2023-07-31:
I tried the latest main:
And still see more of it:

```sh
In file included from ../3rdParty/FreeRTOS-Kernel/include/FreeRTOS.h:64,
                 from ../3rdParty/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/port.c:34:
../3rdParty/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/port.c: In function 'vYieldCore':
../3rdParty/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/port.c:533:26: warning: comparison of integer expressions of different signedness: 'int' and 'uint' {aka 'unsigned int'} [-Wsign-compare]
  533 |     configASSERT(xCoreID != portGET_CORE_ID());
      |                          ^~
../config/jd_rtos_port/FreeRTOSConfig.h:121:54: note: in definition of macro 'configASSERT'
  121 | #define configASSERT(x)                         if(!(x)) jd_assert(__FUNCTION__, __LINE__)
      |                                                      ^
[113/114] Building C object CMakeFiles/JadeDrive.dir/3rdParty/FreeRTOS-Kernel/tasks.c.obj
../3rdParty/FreeRTOS-Kernel/tasks.c: In function 'vTaskGetInfo':
../3rdParty/FreeRTOS-Kernel/tasks.c:5549:40: warning: assignment discards 'volatile' qualifier from pointer target type [-Wdiscarded-qualifiers]
 5549 |             pxTaskStatus->pxTopOfStack = pxTCB->pxTopOfStack;
      |    
```

pranjalchanda08, 2023-07-31:
Would be easier to get to the bottom of it using:
```txt
target_compile_options( ${TARGET} PUBLIC
    ### Gnu/Clang C Options
    $<$<COMPILE_LANG_AND_ID:C,GNU>:-fdiagnostics-color=always>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-fcolor-diagnostics>

    $<$<COMPILE_LANG_AND_ID:C,Clang,GNU>:-Wall>
    $<$<COMPILE_LANG_AND_ID:C,Clang,GNU>:-Wextra>
    # $<$<COMPILE_LANG_AND_ID:C,Clang,GNU>:-Werror>
    $<$<COMPILE_LANG_AND_ID:C,Clang>:-Weverything>

)
```

added to the CmakLists.txt of RPi-pico project

ActoryOu, 2023-07-31:
Thanks for recommendation!
We have CI to compile pico project every PR.
Now I'm trying to apply these options to that project.

pranjalchanda08, 2023-07-31:
Are we planning to have further fixed to these?

pranjalchanda08, 2023-07-31:
> I tried the latest main:
> And still see more of it:
> 
> ```sh
> In file included from ../3rdParty/FreeRTOS-Kernel/include/FreeRTOS.h:64,
>                  from ../3rdParty/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/port.c:34:
> ../3rdParty/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/port.c: In function 'vYieldCore':
> ../3rdParty/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/port.c:533:26: warning: comparison of integer expressions of different signedness: 'int' and 'uint' {aka 'unsigned int'} [-Wsign-compare]
>   533 |     configASSERT(xCoreID != portGET_CORE_ID());
>                                                    ^
> [113/114] Building C object CMakeFiles/JadeDrive.dir/3rdParty/FreeRTOS-Kernel/tasks.c.obj
> ../3rdParty/FreeRTOS-Kernel/tasks.c: In function 'vTaskGetInfo':
> ../3rdParty/FreeRTOS-Kernel/tasks.c:5549:40: warning: assignment discards 'volatile' qualifier from pointer target type [-Wdiscarded-qualifiers]
>  5549 |             pxTaskStatus->pxTopOfStack = pxTCB->pxTopOfStack;
>       |    
> ```
> 

 Are we planning to have further fixes to these?

ActoryOu, 2023-07-31:
Hi @pranjalchanda08,
Sure. I'll fix it with CI change.

ActoryOu, 2023-08-01:
Hi @pranjalchanda08,
Unfortunately I cannot reproduce some warnings in my local environment.
Could you help to provide compile log that you meet so I can resolve them all in one shot?

BTW, I'm using Windows. I need to set up an Ubuntu environment to reproduce it, but it'll take some time.

Thanks.

pranjalchanda08, 2023-08-01:
> Hi @pranjalchanda08, Unfortunately I cannot reproduce some warnings in my local environment. Could you help to provide compile log that you meet so I can resolve them all in one shot?
> 
> I'm using Windows. I need to set up an Ubuntu environment to reproduce it, but it'll take some time.
> 
> Thanks.

Mine is also WSL over Windows environment.
As mentioned before following are my config settings:

```c
#if FREE_RTOS_KERNEL_SMP // set by the RP2040 SMP port of FreeRTOS
/* SMP port only */
#define configNUMBER_OF_CORES                   2
#define configTICK_CORE                         0
#define configRUN_MULTIPLE_PRIORITIES           1
#define configUSE_CORE_AFFINITY                 1
#define configUSE_MINIMAL_IDLE_HOOK             0
#endif

/* RP2040 specific */
#define configSUPPORT_PICO_SYNC_INTEROP         1
#define configSUPPORT_PICO_TIME_INTEROP         1
```

If you can share me the way to sync to your branch I can pull it locally and check so that I can share the build logs right away!

ActoryOu, 2023-08-01:
@pranjalchanda08,

That would be super helpful.
Please help to use [PicoWarn2](https://github.com/ActoryOu/FreeRTOS-Kernel/tree/PicoWarn2) branch.

Thanks a lot!

RichardBarry, 2023-08-01:
Your original post shows an unsigned value being compared with >= 0, which will generate the warning, but I cannot find that line in the source code.  Please provide a link to the line in Github.  For example, this is how to link to line 197 in the same sourcefile:  https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/portable/ThirdParty/GCC/RP2040/include/portmacro.h#L197

pranjalchanda08, 2023-08-01:
> Your original post shows an unsigned value being compared with >= 0, which will generate the warning, but I cannot find that line in the source code. Please provide a link to the line in Github. For [...]

That is already resolved using PR #732, post that some more are getting generated

ActoryOu, 2023-08-02:
Hi @pranjalchanda08,
Thanks for your contribution! That really helps!
#736 is created to fix the following warnings first.
Then I'll create another on FreeRTOS repo to add that into CI flow.

pranjalchanda08, 2023-08-02:
> Hi @pranjalchanda08,
> Thanks for your contribution! That really helps!
> #736 is created to fix the following warnings first.
> After merging this, I'll create another PR on FreeRTOS repo to add this into CI flow.

Thanks @ActoryOu,

For all the spontaneous  and continuous support.

Really appreciate your quick response!

Regards,

pranjalchanda08, 2023-08-03:
I tested the current main branch and build the code. It works!. Thanks
The issue can be closed now.