FreeRTOS/FreeRTOS-Kernel #740: [BUG] xEventGroupBitsFromISR API doesn't work correctly for RL78  MCU

KeitaKashima, 2023-08-04:
I met the issue when using RL78.

xEventGroupBitsFromISR API is not working properly on RL78.

Specifically, it seems that portYIELD_FROM_ISR() does not work correctly.

The correct procedure is "BRK execution -> stack save -> context switch -> stack load", but due to an implementation error, it seems that only the context switch is done. 

Therefore, it cannot return to the original and cannot move.

https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/a5bf4d9a7f5fbbb6110ee329c8517641efb9f563/portable/GCC/RL78/portmacro.h#L109

### Fault (current description)
 `#define portYIELD_FROM_ISR( xHigherPriorityTaskWoken ) if( xHigherPriorityTaskWoken ) vTaskSwitchContext()`

### Countermeasures (Fixed)
The following changes will fix the problem.

 `#define portYIELD_FROM_ISR( x ) if( x ! = pdFALSE ) portYIELD()`

Could you check it?

chinglee-iot, 2023-08-04:
Hi KeitaKashima,

Some of the implementation always save/restore the context when entering/leaving the ISR.
If there yield is called in ISR, the current task TCB is switched and later restored before leaving the ISR.

For example, the RL78 tick ISR implemenation the portSAVE_CONTEXT/portRESTORE_CONTEXT is called.
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/02be485e0452313d347678f32edaf79525de7b9a/portable/GCC/RL78/portasm.S#L66-L78

If you are implementing an ISR handler, can you give this method a try and feedback the result?
```
_YourISR:
    portSAVE_CONTEXT

    /* Handling ISR and calling xxxFromISR API here. */    

    portRESTORE_CONTEXT
    reti
```

KeitaKashima, 2023-08-07:
Thank you for your reply

I already checked fix of the operation using the below method.
The method is adapted to your proposal.

https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/02be485e0452313d347678f32edaf79525de7b9a/portable/GCC/RL78/portasm.S#L44

[image]

chinglee-iot, 2023-08-09:
You are welcome. I will close this issue. Feel free to reopen this issue or create a new one if there is any further question.

KeitaKashima, 2023-08-09:
@chinglee-iot 

I think you should update the below GCC code of RL78 to the fixed code we discussed.

* Fault (current your implementation)

`#define portYIELD_FROM_ISR( xHigherPriorityTaskWoken ) if( xHigherPriorityTaskWoken ) vTaskSwitchContext()`

* Countermeasures (Fixed code)

`#define portYIELD_FROM_ISR( x ) if( x ! = pdFALSE ) portYIELD()`

Actual code Line:

https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/a5bf4d9a7f5fbbb6110ee329c8517641efb9f563/portable/GCC/RL78/portmacro.h#L109

chinglee-iot, 2023-08-09:
@KeitaKashima 

Can you share your ISR implementation which your observe the problem?

The implementation portYIELD_FROM_ISR should work if portSAVE_CONTEXT is called as "stack save" when entering the ISR and portRESTORE_CONTEXT is called as "stack load" when leaving the ISR.

kar-rahul-aws, 2023-08-30:
@KeitaKashima 
Did @chinglee-iot 's suggestion solve your issue, without needing to change the current implementation of portYIELD_FROM_ISR
If not, can you share your ISR implementation where you observe the problem?

KeitaKashima, 2023-08-31:
It needs to be fixed portYIELD_FROM_ISR( x ) as I said.
I will share the project that it happned later.  Please wait for a little while longer.

amazonKamath, 2023-11-29:
@KeitaKashima do you have any updates for us? Thank you.

chinglee-iot, 2024-02-01:
@KeitaKashima 
We will close this issue for now. Feel free to reopen this issue when you have updated information.

closed 2024-02-01 by chinglee-iot