FreeRTOS/FreeRTOS-Kernel #752: [BUG]

wiwocd, 2023-08-15:
**Describe the bug**
On the latest version, using heap_5 applying for memory will result in an illegal address that is not 0.

htibosch, 2023-08-15:
Hi @wiwocd, can you indicate which release you're testing, more pricisely than "the latest version"?

Is that today's [head version](https://github.com/FreeRTOS/freertos)? Or [FreeRTOSv202212.01](https://github.com/FreeRTOS/FreeRTOS/releases/tag/202212.01)?

And do you remember which release seemed to work well for you?

wiwocd, 2023-08-15:
V10.6.0

wiwocd, 2023-08-15:
V10.4.3 works well

RichardBarry, 2023-08-15:
Please provide more detail.  What are the steps needed to replicate this issue?

htibosch, 2023-08-15:
In addition, could you provide us with the code that initialises the heap module?
Especially the code around:
~~~
void vPortDefineHeapRegions( const HeapRegion_t * const pxHeapRegions )
~~~

How do you define `pxHeapRegions`?

And if it is possible, could you send a list of all alloc/free calls to the system? With that, it would be possible to simulate your case.
Thank you

htibosch, 2023-08-16:
@wiwocd wrote:

> V10.4.3 works well
> V10.6.0 showed problems

I have a program that will test heap_x modules. It provides memory from the OS and then it will run a series of tests, calling malloc/free in a random order and with random sizes.

I just linked the testing program with the two modules:

    heap_5_V10.4.3.c
    heap_5_V10.6.0.c

and renamed the public functions to e.g. `pvPortMalloc_V10_6_0()`.

The two versions of `heap_5.c` show exactly the same behaviour and `pvPortMalloc()` did not return any invalid pointer.

Maybe it helps if you can make a list of malloc/free calls:

~~~
    // M = malloc, F = free, followed by a pointer and a size
    M 246d070 393
    M 246db58 113
    F 246db58
    F 246d070
~~~

and show the exact values in your `HeapRegion_t` struct.

Also a copy of your `FreeRTOSConfig.h` could give more insight.

Thanks

wiwocd, 2023-08-19:
@htibosch 
Sorry, it's the problem I used, calling vPortFree repeatedly

wiwocd, 2023-08-19:
There's still a problem, here are some logs：
// M = malloc, F = free
M 0x10006e28 4144
M 0x10006a70 88
M 0x10006448 12
M 0x10006460 19
M 0x10006ad0 177
M 0x10006b90 27
M 0x10006bb8 35
M 0x10006be8 256
F 0x10006448
M 0x10006cf0 256
M 0x10007e60 340
F 0x10007e60
F 0x10006cf0
M 0x10007e60 340
M 0x10007fc0 512
M 0x10006cf0 256
M 0x100081c8 335                        
F 0x10007fc0                            
F 0x10007e60                            
M 0x10007e60 340                        
M 0x10007fc0 512                        
F 0x10007fc0                            
F 0x10007e60                                                                   
F 0x10006cf0
F 0x100081c8
F 0x10006be8
F 0x10006460
F 0x10006ad0
F 0x10006bb8
F 0x10006b90
F 0x10006a70
F 0x10006e28
M 0x10006e28 4144
M 0x10006a70 88
M 0x10006448 8
M 0x10006458 19
M 0x10006a08 177
M 0x10006478 27
M **0x8** 35   --------- Illegal pointer returned here
The starting address of the heap is 0x10000000, and the ending address is 0x1000FFFF.

RichardBarry, 2023-08-19:
Although the address are different, I've tried to replicate the exact sequence you posted above (see the code below), and the pointer returned in the final allocation was not corrupt.  If you are able to replicate this error consistently using the sequence above then please step through the last allocation in the debugger so you can see what goes wrong.

Are you using the memory in between allocations and frees, or literally just using the sequence above without using the memory.  If you are using the memory, then I suspect you are corrupting something - which is why the instructions for opening bug reports say to discuss issues on the support forums first.

```
pvReturnedAddresss[ 0 ] = pvPortMalloc( 4144 );//M 0x10006e28 4144
configASSERT( ( ( uint32_t ) pvReturnedAddresss[ 0 ] > ( uint32_t ) &( ucHeap[ 1 ] ) ) && \
              ( ( uint32_t ) pvReturnedAddresss[ 0 ] < ( uint32_t ) &( ucHeap[ mainREGION_1_SIZE ] ) ) );

pvReturnedAddresss[ 1 ] = pvPortMalloc( 88 );//M 0x10006a70 88
configASSERT( ( ( uint32_t ) pvReturnedAddresss[ 1 ] > ( uint32_t ) &( ucHeap[ 1 ] ) ) && \
              ( ( uint32_t ) pvReturnedAddresss[ 1 ] < ( uint32_t ) &( ucHeap[ mainREGION_1_SIZE ] ) ) );

pvReturnedAddresss[ 2 ] = pvPortMalloc( 12 );//M 0x10006448 12
configASSERT( ( ( uint32_t ) pvReturnedAddresss[ 2 ] > ( uint32_t ) &( ucHeap[ 1 ] ) ) && \
              ( ( uint32_t ) pvReturnedAddresss[ 2 ] < ( uint32_t ) &( ucHeap[ mainREGION_1_SIZE ] ) ) );

pvReturnedAddresss[ 3 ] = pvPortMalloc( 19 );//M 0x10006460 19
configASSERT( ( ( uint32_t ) pvReturnedAddresss[ 3 ] > ( uint32_t ) &( ucHeap[ 1 ] ) ) && \
              ( ( uint32_t ) pvReturnedAddresss[ 3 ] < ( uint32_t ) &( ucHeap[ mainREGION_1_SIZE ] ) ) );

pvReturnedAddresss[ 4 ] = pvPortMalloc( 177 );//M 0x10006ad0 177
configASSERT( ( ( uint32_t ) pvReturnedAddresss[ 4 ] > ( uint32_t ) &( ucHeap[ 1 ] ) ) && \
              ( ( uint32_t ) pvReturnedAddresss[ 4 ] < ( uint32_t ) &( ucHeap[ mainREGION_1_SIZE ] ) ) );

pvReturnedAddresss[ 5 ] = pvPortMalloc( 27 );//M 0x10006b90 27
configASSERT( ( ( uint32_t ) pvReturnedAddresss[ 5 ] > ( uint32_t ) &( ucHeap[ 1 ] ) ) && \
              ( ( uint32_t ) pvReturnedAddresss[ 5 ] < ( uint32_t ) &( ucHeap[ mainREGION_1_SIZE ] ) ) );

pvReturnedAddresss[ 6 ] = pvPortMalloc( 35 );//M 0x10006bb8 35
configASSERT( ( ( uint32_t ) pvReturnedAddresss[ 6 ] > ( uint32_t ) &( ucHeap[ 1 ] ) ) && \
              ( ( uint32_t ) pvReturnedAddresss[ 6 ] < ( uint32_t ) &( ucHeap[ mainREGION_1_SIZE ] ) ) );

pvReturnedAddresss[ 7 ] = pvPortMalloc( 256 );//M 0x10006be8 256
configASSERT( ( ( uint32_t ) pvReturnedAddresss[ 7 ] > ( uint32_t ) &( ucHeap[ 1 ] ) ) && \
              ( ( uint32_t ) pvReturnedAddresss[ 7 ] < ( uint32_t ) &( ucHeap[ mainREGION_1_SIZE ] ) ) );

vPortFree( pvReturnedAddresss[ 2 ] );//F 0x10006448

pvReturnedAddresss[ 8 ] = pvPortMalloc( 256 );//M 0x10006cf0 256
configASSERT( ( ( uint32_t ) pvReturnedAddresss[ 8 ] > ( uint32_t ) &( ucHeap[ 1 ] ) ) && \
              ( ( uint32_t ) pvReturnedAddresss[ 8 ] < ( uint32_t ) &( ucHeap[ mainREGION_1_SIZE ] ) ) );

pvReturnedAddresss[ 9 ] = pvPortMalloc( 340 );//M 0x10007e60 340
configASSERT( ( ( uint32_t ) pvReturnedAddresss[ 9 ] > ( uint32_t ) &( ucHeap[ 1 ] ) ) && \
              ( ( uint32_t ) pvReturnedAddresss[ 9 ] < ( uint32_t ) &( ucHeap[ mainREGION_1_SIZE ] ) ) );

vPortFree( pvReturnedAddresss[ 9 ] );//F 0x10007e60
vPortFree( pvReturnedAddresss[ 8 ] );// 0x10006cf0

pvReturnedAddresss[ 10 ] = pvPortMalloc( 340 );//M 0x10007e60 340
configASSERT( ( ( uint32_t ) pvReturnedAddresss[ 10 ] > ( uint32_t ) &( ucHeap[ 1 ] ) ) && \
              ( ( uint32_t ) pvReturnedAddresss[ 10 ] < ( uint32_t ) &( ucHeap[ mainREGION_1_SIZE ] ) ) );

pvReturnedAddresss[ 11 ] = pvPortMalloc( 512 );//M 0x10007fc0 512
configASSERT( ( ( uint32_t ) pvReturnedAddresss[ 11 ] > ( uint32_t ) &( ucHeap[ 1 ] ) ) && \
              ( ( uint32_t ) pvReturnedAddresss[ 11 ] < ( uint32_t ) &( ucHeap[ mainREGION_1_SIZE ] ) ) );

pvReturnedAddresss[ 12 ] = pvPortMalloc( 256 );//M 0x10006cf0 256
configASSERT( ( ( uint32_t ) pvReturnedAddresss[ 12 ] > ( uint32_t ) &( ucHeap[ 1 ] ) ) && \
              ( ( uint32_t ) pvReturnedAddresss[ 12 ] < ( uint32_t ) &( ucHeap[ mainREGION_1_SIZE ] ) ) );

pvReturnedAddresss[ 13 ] = pvPortMalloc( 335 );//M 0x100081c8 335
configASSERT( ( ( uint32_t ) pvReturnedAddresss[ 13 ] > ( uint32_t ) &( ucHeap[ 1 ] ) ) && \
              ( ( uint32_t ) pvReturnedAddresss[ 13 ] < ( uint32_t ) &( ucHeap[ mainREGION_1_SIZE ] ) ) );

vPortFree( pvReturnedAddresss[ 11 ] );// F 0x10007fc0
vPortFree( pvReturnedAddresss[ 10 ] );// F 0x10007e60

pvReturnedAddresss[ 14 ] = pvPortMalloc( 340 );//M 0x10007e60 340
[... 23 lines cut ...]
pvReturnedAddresss[ 17 ] = pvPortMalloc( 88 );//M 0x10006a70 88
configASSERT( ( ( uint32_t ) pvReturnedAddresss[ 17 ] > ( uint32_t ) &( ucHeap[ 1 ] ) ) && \
              ( ( uint32_t ) pvReturnedAddresss[ 17 ] < ( uint32_t ) &( ucHeap[ mainREGION_1_SIZE ] ) ) );

pvReturnedAddresss[ 18 ] = pvPortMalloc( 8 );//M 0x10006448 8
configASSERT( ( ( uint32_t ) pvReturnedAddresss[ 18 ] > ( uint32_t ) &( ucHeap[ 1 ] ) ) && \
              ( ( uint32_t ) pvReturnedAddresss[ 18 ] < ( uint32_t ) &( ucHeap[ mainREGION_1_SIZE ] ) ) );

pvReturnedAddresss[ 19 ] = pvPortMalloc( 19 );//M 0x10006458 19
configASSERT( ( ( uint32_t ) pvReturnedAddresss[ 19 ] > ( uint32_t ) &( ucHeap[ 1 ] ) ) && \
[... cut ...]
pvReturnedAddresss[ 21 ] = pvPortMalloc( 27 );//M 0x10006478 27
configASSERT( ( ( uint32_t ) pvReturnedAddresss[ 21 ] > ( uint32_t ) &( ucHeap[ 1 ] ) ) && \
              ( ( uint32_t ) pvReturnedAddresss[ 21 ] < ( uint32_t ) &( ucHeap[ mainREGION_1_SIZE ] ) ) );

pvReturnedAddresss[ 22 ] = pvPortMalloc( 8 );
configASSERT( ( ( uint32_t ) pvReturnedAddresss[ 22 ] > ( uint32_t ) &( ucHeap[ 1 ] ) ) && \
              ( ( uint32_t ) pvReturnedAddresss[ 22 ] < ( uint32_t ) &( ucHeap[ mainREGION_1_SIZE ] ) ) );

```

htibosch, 2023-08-20:
I have the same results here: all allocations and calls to `vPortFree()` are as expected.

What we've done so far:

1. I proved earlier that the two versions of `heap_5.c` in V10.4.3 and V10.6.0 work exactly the same.
2. @RichardBarry showed that the order of allocations works perfectly without triggering an assert.

It looks like it is your turn: find the code that overwrites or corrupts the allocated memory.

What I sometimes did is place an allocator in between the application and heap_5. It will allocate a few byte bytes extra:

~~~
    unsigned length;
    unsigned signature_1;
    uint8_t allocated_memory[]; // variable length
    uint8_t signature_2[ 4 ];
~~~

When freeing a block of memory, the de-allocator should find the two signatures, and also the length byte.
If anything is wrong, it would raise an assert.

Maybe the next time you can open a case on the [FreeRTOS forum](https://forums.freertos.org) in stead of Github.

wiwocd, 2023-08-22:
**@htibosch @RichardBarry**   Sorry, I'm checking my own code.

htibosch, 2023-08-22:
Thanks for letting us know.

I think that you can close the issue for now:

[image]

And as I said, you can always ask questions on the FreeRTOS forum.

wiwocd, 2023-08-28:
This issue is that strcpy will be copied together with '\0', causing memory to exceed the limit.
Thanks all.

closed 2023-08-28 by wiwocd