FreeRTOS/FreeRTOS-Kernel #767: [BUG] ARM_CM23_nonsecure does not support psplim

Dasheverless, 2023-08-30:
**Describe the bug**
  Here is the link of [Arm Cortex-M23 Devices Generic User Guide](https://developer.arm.com/documentation/dui1095/a/The-Cortex-M23-Processor/Programmers-model/Core-registers?lang=en). As you can see, in the table 2.3, there is no psplim register in Non-secure state. Im not sure what will happen in a cm23 board, cause I only run the code on vm. If I delete the line that uses psplim, the code will work.
**Official code**
Several lines in portasm.s use psplim.
[msr psplim, r3](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/af2904b01c243b5d59ea477fc8564679b451fcd8/portable/GCC/ARM_CM23_NTZ/non_secure/portasm.c#L112)
[msr  psplim, r1](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/af2904b01c243b5d59ea477fc8564679b451fcd8/portable/GCC/ARM_CM23_NTZ/non_secure/portasm.c#L158)

urutva, 2023-08-30:
@Dasheverless Apologies, I pressed the enter before completing the response.

Can you please provide more information on the example you are running and on which platform?

I'll look into this and create a PR soon.

Dasheverless, 2023-08-30:
Hi @urutva ,
The architecture manual says when access psplim in non secure state with base extension, it will lead RAZ/WI. However, it reports a hard fault when reading psplim while nothing happened when writting psplim on our inhouse vm. Maybe it is a vm bug. If you are familiar with the behavior of msr or mrs a raz/wi register, please let me know.
Thanks!

urutva, 2023-09-01:
Hi @Dasheverless, apologies for late reply.

From the From the [Armv8-M architecture manual](https://developer.arm.com/documentation/ddi0553/bw/?lang=en), section `D1.2.217 PSPLIM, Process Stack Pointer Limit Register` and `Configuration` sub-section described that `This register is always implemented.`. However, If the Main Extension is not implemented, the Non-secure PSPLIM
is RES0.

Dasheverless, 2023-09-19:
Hi @urutva , someone meet the same issue but in a board not in vm. So, I consult the processor team, they told me this.
You can see the **Operation for all encodings** section in C2.4.125 MRS and C2.4.126 MSR of [Armv8-M architecture manual](https://developer.arm.com/documentation/ddi0553/bw/?lang=en).
    For example, when using cm23 witout security extension, it is always in non-secure state. The SYSm of PSPLIM_NS is 10001011.
 ```
case SYSm[7:3] of
...
    when '10001'
        if !HaveSecurityExt() then UNPREDICTABLE;
 ```
As you can see, when access PSPLIM in a baseline and wiout security extension, the behaviour is UNPREDICTABLE which may cause hardfualt.

jefftenney, 2023-09-20:
@Dasheverless A minor correction -- but no change to the conclusion.  FreeRTOS code never uses the 10001011 encoding, because FreeRTOS code never explicitly addresses PSPLIM_NS.  Instead, the code uses PSPLIM (no suffix), which gets the 00001011 encoding.  That's important because the operation for that encoding considers the main extension when the PE does not have the security extension. That way FreeRTOS can still use the PSPLIM register with the main extension and without the security extension.

[image]

Dasheverless, 2023-09-20:
> @Dasheverless A minor correction -- but no change to the conclusion. FreeRTOS code never uses the 10001011 encoding, because FreeRTOS code never explicitly addresses PSPLIM_NS. Instead, the code [...]

Thanks for correction!

jefftenney, 2023-09-20:
@Dasheverless Oh I meant to ask, which board encountered this error?  (Which MCU model, vendor, etc.)

Dasheverless, 2023-09-20:
> @Dasheverless Oh I meant to ask, which board encountered this error? (Which MCU model, vendor, etc.)

It is a inhouse board, similar to cm23.

aggarg, 2023-09-28:
Closing this issue as the following PR https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/791 is merged.

closed 2023-09-28 by aggarg