FreeRTOS/FreeRTOS-Kernel #792: [BUG] xStreamBufferCreateStatic() fails when xBufferSizeBytes is 1.

anolivetree, 2023-09-12:
**Describe the bug**
xStreamBufferCreateStatic() fails when xBufferSizeBytes is 1.

**To Reproduce**
- call xStreamBufferCreateStatic() with xBufferSizeBytes 1

**Expected behavior**
- creates stream buffer successfully.

**Additional context**
xStreamBufferGenericCreateStatic do following check even when xIsMessageBuffer is pdFALSE.

> configASSERT( xBufferSizeBytes > sbBYTES_TO_STORE_MESSAGE_LENGTH );

It seems the check should be executed only when xIsMessageBuffer is pdTRUE, just like xStreamBufferGenericCreate() does.

kar-rahul-aws, 2023-09-13:
Hi @anolivetree 
Thank you for reporting this issue. I have created a patch for the fix, and it has been merged to the mainline. Please let us know if this fixes the issue you were observing.
Thanks.

kar-rahul-aws, 2023-09-13:
Sure, thank you for the update.
Closing the issue.

closed 2023-09-13 by kar-rahul-aws