FreeRTOS/FreeRTOS-Kernel #811: [Feature Request] Add API to get task base priority.

Spamfast, 2023-09-26:
We use FreeRTOS a lot but come up against a snag when implementing temporary task priority boosting functionality.

The problem is that when mutexes are enabled in FreeRTOSConfig.h the value set by vTaskPrioritySet() is the base priority but the value returned by uxTaskPriorityGet() is the dynamic priority which may not be the same if the task currently owns a mutex that is waited on by a higher priority thread. So a task can't get its base priority, change it for a while and then restore it.

We work around this using vTaskGetInfo() to implement a uxTaskBasePriorityGet() but that is very inefficient.

It would be far better if the functionality were implemented in task.c and eposed in task.h.

chinglee-iot, 2023-09-27:
Thank you for the suggestion. We would discuss about this API and later reply in this thread.

kar-rahul-aws, 2023-10-17:
Closing the issue, since the fix has been merged in the [PR](https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/818).

closed 2023-10-17 by kar-rahul-aws