FreeRTOS/FreeRTOS-Kernel #854: [Feature Request] Add portMEMORY_BARRIER definition for Renesas RX MCU

YuichiNakadaSX, 2023-10-23:
We have confirmed that the demo does not work correctly when optimizing with the compiler on Renesas RX MCU.
The PortMEMORY_BARRIER definition must be added to portmacro.h.

### Confirmed phenomenon
The demo does not work correctly under the following conditions:
- Project: https://github.com/FreeRTOS/FreeRTOS/tree/main/FreeRTOS/Demo/RX700_RX72N_EnvisionKit_Renesas_e2studio_CS%2B
- Set compiler optimization to "-Omax -speed".

When it runs the demo, it loops infinitely in xTaskResumeAll in tasks.c.

### Solution (request for this issue)
Add portMEMORY_BARRIER definition to portmacro.h as follows:
- GCC
  ```c
  #define portMEMORY_BARRIER()    __asm volatile ( "" ::: "memory" )
  ```
- CC-RX
  ```c
  #pragma inline_asm vPortMemoryBarrier
  static void vPortMemoryBarrier( void )
  {
  }
  
  #define portMEMORY_BARRIER()    vPortMemoryBarrier()
  ```

### Target
All series of Renesas RX.

ActoryOu, 2023-10-24:
Hi @YuichiNakadaSX,
Thanks for reporting this to us. We're working on the solution.
Will let you know here.

YuichiNakadaSX, 2023-10-24:
Hi @ActoryOu,
Thank you for your comment.
I am sorry that I am not able to PR about this because I need some paperwork if I do PR.
Could you please update it on your side?

Skptak, 2023-10-26:
Hey @YuichiNakadaSX, can you please validate if #864 is the exact set of changes you are looking for?

Thanks

TakeoTakahashi2020, 2023-11-01:
Hi @Skptak, thank you very much for your PR.  I have confirmed your #864 code on @YuichiNakadaSX  behalf.

* Using 10.6.1 kernel with your portmacro.h for RX MCU (RX100/RX600v2/RX700v3_DPFPU) of GCC & Renesas, this issue was solved, and other portmacros.h for RX MCU also looks good. 
* Please remove change for Renesas/SH2A_FPU because it is not RX MCU.

BTW, inline assemble code in your portmacro.h had to be corrected which caused compile errror.  The inline assemble code format  seems to be broken by uncrustify.  Each port.c in portable/Renesas/RX* also has same issue.

Could you please fix the inline assemble code and add `/* *INDENT-OFF* */` and `/* *INDENT-ON* */ ` as below to avoid compile errror issue.  

[image]

Left code is your original code.  Right code is fixed and INDENT-OFF/INDENT-ON is added.

Skptak, 2023-11-03:
Hey @TakeoTakahashi2020 
I've raised #876 to fix the formatting issues
Once that is approved and merged I'll fix the changes on this PR

Thanks

TakeoTakahashi2020, 2023-11-06:
Hi @Skptak san,   

I have confirmed #876.   Thank you very much.

> Once that is approved and merged I'll fix the changes on this PR

Understood.  

Thanks.

Skptak, 2023-11-29:
Hey @TakeoTakahashi2020, #864  has been merged into the  mainline kernel, I'm going to mark this issue as closed now
Please feel free to create a new issue if you need anything else!

closed 2023-11-29 by Skptak

TakeoTakahashi2020, 2023-11-30:
Hi @Skptak san,
I confirmecd that #876 and #864 have been merged.
Thank you very much!