FreeRTOS/FreeRTOS-Kernel #904: FreeRTOS-Kernel V11 - SMP

amazonKamath, 2023-11-27:
Symmetric Multiprocessing (SMP) support in the FreeRTOS Kernel enables one instance of the FreeRTOS kernel to schedule tasks across multiple identical processor cores. Currently, SMP support exists in a separate branch. This work item involves releasing the FreeRTOS SMP support as Generally Available and have a common codebase for single core and SMP.

sirnish, 2023-12-18:
Please read more [here](https://freertos.org/2023/12/introducing-freertos-kernel-version-11-0-0-a-major-release-with-symmetric-multiprocessing-smp-support.html)

sirnish, 2023-12-18:
Please see more [here](https://freertos.org/2023/12/introducing-freertos-kernel-version-11-0-0-a-major-release-with-symmetric-multiprocessing-smp-support.html)

closed 2023-12-18 by sirnish

aggarg, 2024-11-28:
Thank you for pointing. I have updated the link.