FreeRTOS/FreeRTOS-Kernel #905: Granular Lock for SMP

amazonKamath, 2023-11-27:
Current FreeRTOS SMP uses a single lsingle set of locks to guard all critical sections and scheduler suspension. This results in lower performance because of lock contention. This work item involves making the locking more granular by having one lock per kernel object to remove lock contention when different cores are not accessing the same kernel object.

Dazza0, 2023-12-11:
@aggarg I can't assign this issue to myself. I think I'll need [collaborator](https://docs.github.com/en/account-and-profile/setting-up-and-managing-your-personal-account-on-github/managing-access-to-your-personal-repositories/inviting-collaborators-to-a-personal-repository) access to be able to do that (cc @sudeep-mohanty in case you need to assign FreeRTOS kernel issues as well).

Dazza0, 2024-01-08:
Note: I'm already working on this issue.