FreeRTOS/FreeRTOS-Kernel #928: SPDX file contains numerous issues

Malcolmnixon, 2023-12-19:
**Describe the issue**
When using the accompanying SPDX file provided with releases, numerous issues. The file seems to be malformed according to the SPDX specification. These issuescan be observed by uploading the files to the SPDX organizations [online validator tool](https://tools.spdx.org/app/validate/).

Note: I assume this falls under "documentation" as the SPDX file is used to document the contents and version information for a release.

**Screenshot**
[image]

kstribrnAmzn, 2023-12-20:
Thank you for bringing this up! I took a look through the validator error and I believe the issues is due to the use of the underline character (aka '_') in the paths which in turn are used in the SPDXID. I'll need to dig a little deeper into this and see if we can update our SPDX tooling to convert these to hypens which are valid per the regex.

Malcolmnixon, 2023-12-20:
Indeed the specification at https://spdx.github.io/spdx-spec/v2.2.2/file-information/#821-description does not seem to permit underscores. We're using https://github.com/microsoft/sbom-tool to generate our SPDX SBOMs and it translates underscores to dashes.

The sbom-tool implementation at https://github.com/microsoft/sbom-tool/blob/main/src/Microsoft.Sbom.Parsers.Spdx22SbomParser/Utils/SPDXExtensions.cs#L38 appears to:
1. Construct the SPDXID as `SPDXRef-File-{fileName}-{sha1Value}`
2. Regex replace anything not valid (E.G. matching `[^a-zA-Z0-9.-]`) with a dash `-`

kstribrnAmzn, 2023-12-20:
Thank you for all of this information. To get this fixed quickly I've updated out tool to build a valid SPDX formatted SBOM. I'm working with my teammates to get this released ASAP.

kstribrnAmzn, 2023-12-20:
@Malcolmnixon we're having a bit of a debate here and would love you input. While we have fixed the file we're discussing the most transparent way to release this and would love your input.

Would you expect to see a version patch increase with the fixed SBOM? Or, given this is metadata, rather see a re-release of FreeRTOS 11.0.0?

Malcolmnixon, 2023-12-20:
If it were up to me I would just wait and include it in the next 11.x release you make. Statistically you seem to be doing releases every 3-4 months. My team is not using 11.0.0 yet, and by the time we upgrade you'll almost certainly have a new release out including the clean SBOMs so we'll just jump to one of those versions.

That being said there may be others out there with a more pressing need to have NTIA compliant SBOMs for all their components.

kar-rahul-aws, 2023-12-21:
[V11.0.1 ](https://github.com/FreeRTOS/FreeRTOS-Kernel/releases/tag/V11.0.1) is released with the updated SBOM file.

closed 2023-12-21 by kar-rahul-aws