FreeRTOS/FreeRTOS-Kernel #955: [BUG] GCC_ARM_CM4_MPU assumes FPU

IsaacDynamo, 2024-01-16:
Building `GCC_ARM_CM4_MPU` with `arm-none-eabi-gcc` fails with `Error: selected FPU does not support instruction`.

**Expected behavior**
Given that the **F** in `GCC_ARM_CM4F` means with FPU, I expected `GCC_ARM_CM4_MPU` to not need a FPU.

But it seems that the `GCC_ARM_CM4_MPU` port expects to have an FPU. So is this a naming inconsistency? And should the port be renamed to `GCC_ARM_CM4F_MPU`?

jasonpcarroll, 2024-01-17:
Hi @IsaacDynamo,

Thanks for bringing this discrepancy to our attention. I will notify the team and see what they think.

Best,

Jason Carroll

jasonpcarroll, 2024-01-23:
Hi @IsaacDynamo,

While I agree that the F in the name represents the use of the FPU, at this point, its probably not worth renaming and causing someone's build to fail or break links. I will close this issue.

Best,

Jason Carroll

closed 2024-01-23 by jasonpcarroll