FreeRTOS/FreeRTOS-Kernel #993: [Feature Request] Default Affinity Set

go2sh, 2024-02-14:
**Is your feature request related to a problem? Please describe.**
We created a freeRTOS SMP port for the Infineon TC3x and TC4x family, but the system design is not really symmetrical. Especially the Task context ist bound to the CPU executing the task. This makes switching tasks between different CPUs nearly impossible. But the SMP design still makes some sense to distribute work across cores more easily. For example in conjunction with FreeRTOS-Plus-TCP to distribute sockets across cores.

**Describe the solution you'd like**
Create a new config flag to allow setting a default affinity for a task, where no affinity set is provided during task creation. This is especially useful for legacy code. This could either be a fixed value or the current core, who creates the task.

**Describe alternatives you've considered**
Non

**How many devices will this feature impact?**
All TC3x and TC4x devices with SMP enabled.

**What are your project timelines?**
A patch is already prepared internally.

go2sh, 2024-02-14:
Would you accept a PR for this?

rawalexe, 2024-02-14:
Hello thank you so much for reporting about this problem, we can definitely consider accepting the PR, Please feel free to go ahead and create your PR, the team can have look at it and share more details on it.

PR #996 (2024-02-16): Add default core affinity config value.

Description
-----------
Add a config define to set the default core affinity mask for tasks created without an affinity mask specified via arguments.

Test Steps
-----------
Set configTASK_DEFAULT_CORE_AFFINITY to 1 and check if all tasks created without an affinity mask specified run on core 0.

Checklist:
----------

- [ x ] I have tested my changes. No regression in existing tests.
- [ ] I have modified and/or added unit-tests to cover the code changes in this Pull Request.

Related Issue
-----------
Closes #993.

By submitting this pull request, I c [... cut]
closed 2024-02-16 by aggarg