FreeRTOS/FreeRTOS-Kernel #1014: Five heaps to choose?

svenbieg, 2024-03-24:
I'm missing my heap in every real-time operating system.

Why do You have five? Fast at middle size, slow at middle size, moderate at average speed, slow but small or fast and big?

Please have a look at my memory-manger at https://github.com/svenbieg/Heap!

Why not copy mine?

Best regards,

Sven Bieg

ActoryOu, 2024-03-25:
Hi @svenbieg,
Thanks for the proposal!
[Memory Management](https://www.freertos.org/a00111.html) is the page introducing different heap methods. In summary:
- [heap_1](https://www.freertos.org/a00111.html#heap_1) - the very simplest, does not permit memory to be freed.
- [heap_2](https://www.freertos.org/a00111.html#heap_2) - permits memory to be freed, but does not coalescence adjacent free blocks.
- [heap_3](https://www.freertos.org/a00111.html#heap_3) - simply wraps the standard malloc() and free() for thread safety.
- [heap_4](https://www.freertos.org/a00111.html#heap_4) - coalescences adjacent free blocks to avoid fragmentation. Includes absolute address placement option.
- [heap_5](https://www.freertos.org/a00111.html#heap_5) - as per heap_4, with the ability to span the heap across multiple non-adjacent memory areas.

IMHO, these methods are indeed employed in various scenarios. The choice between them ultimately depends on the specific use case. I'll bring this into our discussion.

Thank you.

ActoryOu, 2024-03-25:
BTW, could you help draft a post introducing this memory management proposal in [FreeRTOS forum](https://forums.freertos.org/c/kernel/5) for broader community?

Thank you.

closed 2024-03-25 by RichardBarry

svenbieg, 2024-03-25:
I'm not interested in FreeRTOS, i want my self-driving car. Tesla is working with custom Linux which will never work.

svenbieg, 2024-03-25:
Allocations and deletions are done in constant low time, fragmentation is no problem. It is also not very big.
This is possible by creating pyramidal directories, sorting gaps by size and offset.
This should fullfill every need.
Please checkout GitHub to learn more!