FreeRTOS/FreeRTOS-Kernel #1015: [Feature Request] The number of system calls needs to support the sizeof（uintptr_t）

snikeguo, 2024-03-26:
**Is your feature request related to a problem? Please describe.**
I need to add a custom SystemCall instruction to execute a custom function.
However, the maximum number of system call instructions is 255.

**Describe the solution you'd like**
```
....
#define SYSTEM_CALL_xUartSend   10001 
....
PRIVILEGED_DATA UBaseType_t uxSystemCallImplementations[ NUM_SYSTEM_CALLS ] =
{
 ...
 xUartSendImpl,
}

...
BaseType_t MPU_xUartSend(..... )
{
__asm volatile
(
    " .syntax unified                                       \n"
                " .extern MPU_xUartSendImpl                            \n"
                "                                                       \n"
                " push {r0}                                             \n"
                " mrs r0, control                                       \n"
                " tst r0, #1                                            \n"
                " pop {r0}                                              \n"
                " bne MPU_xUartSend_Unpriv                             \n"
                " MPU_xUartSend_Priv:                                  \n"
                "     b MPU_xUartSendImpl                              \n"
                " MPU_xUartSend_Unpriv                                 \n"
                " push SYSTEM_CALL_xUartSend "  //push user system call id into stack
                "     svc %0                                            \n"
                "                                                       \n"
                : : "i" ( SYSTEM_CALL_User ) : "memory"
);
} 
...
```

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**How many devices will this feature impact?**
Expected volume for your product.

**What are your project timelines?**
Timeline for milestones such as design completion, testing and validation, and production.

**Additional context**
Add any other context or screenshots about the feature request here.

If you have the same (or similar) feature request, please upvote this issue with thumbs up 👍
and use the comments section to provide answers to the questions above.

Skptak, 2024-03-26:
What port are you using?
The ARMv7-M and ARMv8-M architectures only support an 8 bit SVC value.
Link to the [ARMv7M Architecture doc](https://developer.arm.com/documentation/ddi0403/d/Application-Level-Architecture/Instruction-Details/Alphabetical-list-of-ARMv7-M-Thumb-instructions/SVC?lang=en)
Link to the [ARMv8M Architecture Download page](https://developer.arm.com/documentation/ddi0553/bx/?lang=en), the relevant section is C2.4.242 SVC
[image]

As this is a Hardware Limitation this request isn't a feature that FreeRTOS can support

snikeguo, 2024-03-26:
@Skptak I have made changes to the implementation. Please refresh the page to view my ideas.

Skptak, 2024-03-26:
Your proposed change is a larger design change, and more of a feature request. Can you open a post on the [FreeRTOS Forums](https://forums.freertos.org/) to discuss it there please?

aggarg, 2024-03-26:
As @Skptak suggested, please open a forum issue. I am closing this one.

closed 2024-03-26 by aggarg