FreeRTOS/FreeRTOS-Kernel #1057: [BUG] Violation of MISRA2012  rule 17.11 - A function that never returns should be declared with a _Noreturn function specifier

markhermeling, 2024-05-14:
**Describe the bug**
In the Coverity example configuration there are 4 places where a function is used that does not return. See the attached SARIF file from a run with CodeSecure CodeSonar. This is a violation of MISRA 17.11, though admittedly not a key rule.

**Target**
- coverity UP and SMP configurations

**Host**
- Linux

**To Reproduce**
- run codesonar analyze -preset misra2012 on project

**Expected behavior**
The specified functions should be defined with _Noreturn, or rule 17.11 should be added as a deviation. I am happy to provide a PR with either of these done, but wanted to file the issue first to understand what the project prefers.

I am working on a few more MISRA violation as it seems that Coverity missed a few rules. Once done, I plan to submit a codesonar example project as well to make it easier for users to run CodeSonar on FreeRTOS.

[Misra17-11-violations.sarif.zip](https://github.com/FreeRTOS/FreeRTOS-Kernel/files/15310607/Misra17-11-violations.sarif.zip)

kar-rahul-aws, 2024-05-15:
Hi @markhermeling 
Thank you for reporting the issue.
We are having a discussion with the team, on the proper approach to address the issue for all ports. 
We will reply back shortly, with the proposed changes, if any.

Thanks.

markhermeling, 2024-05-16:
> We are having a discussion with the team, on the proper approach to address the issue for all ports. 

I have another couple of issues to report. Looking for suggestions from the team as to how to approach. 

What I would like to do is to submit a codesonar example and make a PR with that and suggested code changes if that is easiest. 

I can also add CodeSonar either to the pipeline in the repo, or run it once per day and publish the results.

n9wxu, 2024-05-17:
We recently started running sonarcube experimentally but it is not tied to PR checks and we may have different configurations.  The easiest path may be submitting a PR with a test run and your changes.  Then we can directly comment on the proposed changes and discuss any configuration differences.

n9wxu, 2024-05-17:
Looks like the kernel is doing sonar checks in the PR.  This PR shows the result.
https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/353

kar-rahul-aws, 2024-06-10:
We have linked Code Sonar here - https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/.github/third_party_tools.md.

closed 2024-06-10 by kar-rahul-aws