FreeRTOS/FreeRTOS-Kernel #1058: [BUG] Violation of MISRA 2012 Rule 5.5: Identifiers shall be distinct from macro names

markhermeling, 2024-05-14:
**Describe the bug**
in include/semphr.h, line 675:
#define xSemaphoreTakeFromISR( xSemaphore, pxHigherPriorityTaskWoken )    xQueueReceiveFromISR( ( QueueHandle_t ) ( xSemaphore ), NULL, ( pxHigherPriorityTaskWoken ) )

In the same file, line 26:
#ifdef xSemaphoreTakeFromISR
 #hard_undef xSemaphoreTakeFromISR
BaseType_t xSemaphoreTakeFromISR( SemaphoreHandle_t xSemaphore, BaseType_t * const pxHigherPriorityTaskWoken )

The macro and the function have the same name, which is a violation of rule 5.5

The same happens to xSemaphoreGive at line 460 and 23

**Target**
- UP and SMP configurations of the Coverity example

**Host**
- Linux

**To Reproduce**
Run codesonar analyze -preset misra2012 on the Coverity example

**Expected behavior**
Either add a deviation, or change the code to have none distinct names for the macro and the function

**Screenshots**
SARIF file attached

[MISRA-violation-5.5.sarif.zip](https://github.com/FreeRTOS/FreeRTOS-Kernel/files/15313402/MISRA-violation-5.5.sarif.zip)

RichardBarry, 2024-05-15:
Hi, thanks for this.  I would be grateful if you can post links to the referenced lines as [line 26 in semphr.h](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/include/semphr.h#L26) appears to be a comment.

kar-rahul-aws, 2024-05-15:
Hi @markhermeling 
Thanks for reporting the issue.
We only see the function prototypes in the comments as pointed out by @RichardBarry .
This would likely not cause an issue, since the macro definitions are proper in the source include file.
Could you please provide more information on the lines( which are not comments ) which have similar issue.

Thanks.

markhermeling, 2024-05-15:
Ah, I see what is going on. Let me retract this bug. This configuration of FreeRTOS used a #define for xSemaphoreTakeFromISR, where the CodeSonar model for FreeRTOS is expecting a function and the two are clashing.

This is not a problem in FreeRTOS, sorry for the noice

closed 2024-05-15 by markhermeling