FreeRTOS/FreeRTOS-Kernel #1063: [DOC] xQueueOverwrite Queue Length

schilkp, 2024-05-17:
**Describe the issue**

The documentation for `xQueueOverwrite` states:
```
xQueueOverwrite() is intended for use with queues that have a length of one, 
meaning the queue is either empty or full.
```

`xQueueOverwrite` calls `xQueueGenericSend`:
```c
#define xQueueOverwrite( xQueue, pvItemToQueue ) \
    xQueueGenericSend( ( xQueue ), ( pvItemToQueue ), 0, queueOVERWRITE )
```

`xQueueGenericSend` features the following assertion:
```c
configASSERT( !( ( xCopyPosition == queueOVERWRITE ) && ( pxQueue->uxLength != 1 ) ) );
```

This means that, that either:
  -  `xQueueOverwrite` is not just intended for queues of length one, but **requires** queues of length one.
  -  The assertion in `xQueueGenericSend` is overly conservative.

**Reference**
[https://www.freertos.org/xQueueOverwrite.html](https://www.freertos.org/xQueueOverwrite.html)

**Browser**
- Browser: [e.g. Chrome]
- Version: [e.g. 80.0.3987.132]

Thanks!

kar-rahul-aws, 2024-05-29:
Hi @schilkp 
Thank you for reporting the issue. We will fix the documentation and update.

kar-rahul-aws, 2024-06-29:
Hi @schilkp 
It has been updated in the website. Closing the issue.

closed 2024-06-29 by kar-rahul-aws