FreeRTOS/FreeRTOS-Kernel #1073: FreeRTOS posix port hangs when issuing interrupt-safe calls from an external thread

doppioandante, 2024-05-25:
FreeRTOS: version: v11.1.0
FreeRTOS port: FreeRTOS Posix port
OS: Ubuntu 22.04

I'm using the freertos posix port to test a the firmware logic without the need for a hardware. The real hardware receives data via an interrupt, which is then pushed on a queue like:
```c
xQueueSendFromISR(..., &waked)
portYIELD_FROM_ISR(waked);
```
On the simulator I'm using an external thread (pthread) not managed by FreeRTOS that wait for events on a local network interface; when an event comes, the same code is called, but I've found that it makes FreeRTOS hang or even assert often.

I also added the following assert to `vPortExitCritical`, and it somestimes get triggered, suggesting me that either there's something wrong in the port or, more probably, I'm invalidating the invariant by using the API incorrectly:
```c
void vPortExitCritical( void )
{
    uxCriticalNesting--;
    configASSERT(uxCriticalNesting >= 0);

    /* If we have reached 0 then re-enable the interrupts. */
    if( uxCriticalNesting == 0 )
    {
        vPortEnableInterrupts();
    }
}
```

(BTW, wouldn't a mutex or at least an atomic increment/decrement be needed for `uxCriticalNesting`?)

I stripped down the pattern that makes my code crash down to:
<summary>main.c</summary>
```c
#define _GNU_SOURCE

#include <FreeRTOS.h>
#include <task.h>
#include <queue.h>

#include <stdio.h>
#include <stdbool.h>
#include <pthread.h>
#include <unistd.h>
#include <signal.h>

static void init_globals();
static void init_tasks();

static void reader_task(void*);
static void periodic_task(void*);
static void* interrupt_simulator(void*);

static void send_int(int value);

static QueueHandle_t queue;

int main() {
    init_globals();
    init_tasks();

    vTaskStartScheduler();
    return 0;
}

void init_globals() {
    queue = xQueueCreate(128, sizeof(int));
}

void init_tasks() {
    xTaskCreate(reader_task, "ReaderTask", configMINIMAL_STACK_SIZE, NULL, 1, NULL);
    xTaskCreate(periodic_task, "PeriodicTask", configMINIMAL_STACK_SIZE, NULL, 2, NULL);

[... 416 lines cut ...]

/* Prototype for the function used to print out.  In this case it prints to the
 * console before the network is connected then a UDP port after the network has
 * connected. */
extern void vLoggingPrintf( const char * pcFormatString,
                            ... );

/* Set to 1 to print out debug messages.  If ipconfigHAS_DEBUG_PRINTF is set to
 * 1 then FreeRTOS_debug_printf should be defined to the function used to print
 * out the debugging messages. */
#define ipconfigHAS_DEBUG_PRINTF    1
#if ( ipconfigHAS_DEBUG_PRINTF == 1 )
    #define FreeRTOS_debug_printf( X )    vLoggingPrintf X
#endif

/* Set to 1 to print out non debugging messages, for example the output of the
 * FreeRTOS_netstat() command, and ping replies.  If ipconfigHAS_PRINTF is set to 1
 * then FreeRTOS_printf should be set to the function used to print out the
 * messages. */
#define ipconfigHAS_PRINTF    0
#if ( ipconfigHAS_PRINTF == 1 )
    #define FreeRTOS_printf( X )    vLoggingPrintf X
#endif
#endif /* FREERTOS_CONFIG_H */
```

paulbartell, 2024-05-29:
@doppioandante Consider using the event api defined in [wait_for_event.c](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/portable/ThirdParty/GCC/Posix/utils/wait_for_event.c) to simulate your interrupt.

shubnil, 2024-06-05:
@doppioandante Please update your observations after using "wait_for_event.c".

doppioandante, 2024-06-05:
I'm not quite sure how I could use that API, it seems to me like an internal implementation detail of the POSIX port.
An "simulated interrupt" mimicking the hardware counterpart would interrupt the current task, do its things and then yield back to freertos, right?
While the `wait_for_event` API looks like it is designed to wait for on a condition, and it's indeed used by the port to resume tasks.

aggarg, 2024-09-04:
What about using a signal handler for simulating interrupt? Something like the following - 
```c
#define _GNU_SOURCE

#include <FreeRTOS.h>
#include <task.h>
#include <queue.h>

#include <stdio.h>
#include <stdbool.h>
#include <pthread.h>
#include <unistd.h>
#include <signal.h>

static void init_globals();
static void init_tasks();

static void reader_task(void*);
static void periodic_task(void*);
static void* interrupt_simulator(void*);

static void send_int(int value);

static QueueHandle_t queue;
static pthread_t tid;

int main() {
    init_globals();
    init_tasks();

    vTaskStartScheduler();
    return 0;
}

void init_globals() {
    queue = xQueueCreate(128, sizeof(int));
}

void sigusr1_handler(int sig) {
    (void) sig;
    static int val = 1;
    BaseType_t waked = pdFALSE;

    xQueueSendFromISR(queue, &val, &waked);
    val += 1;
    portYIELD_FROM_ISR(waked);
}

void init_tasks() {
    xTaskCreate(reader_task, "ReaderTask", configMINIMAL_STACK_SIZE, NULL, 1, NULL);
    xTaskCreate(periodic_task, "PeriodicTask", configMINIMAL_STACK_SIZE, NULL, 2, NULL);

    sigset_t set;
    pthread_attr_t attr;
    struct sigaction sigusr;

    sigfillset(&set);
    sigdelset( &set, SIGUSR1 );
    pthread_attr_init(&attr);
    pthread_attr_setsigmask_np(&attr, &set);

    sigusr.sa_flags = 0;
    sigusr.sa_handler = sigusr1_handler;
    sigfillset( &sigusr.sa_mask );
    sigaction( SIGUSR1, &sigusr, NULL );

    pthread_create(&tid, &attr, interrupt_simulator, NULL);
}

void reader_task(void* args) {
    (void) args;

    while (true) {
        int val;
        xQueueReceive(queue, &val, portMAX_DELAY);

        send_int(val+1);
    }
}

void periodic_task(void* args) {
    (void) args;

    TickType_t previousWakeTime = xTaskGetTickCount();

    while (true) {
        send_int(0);

        xTaskDelayUntil(&previousWakeTime, pdMS_TO_TICKS(100));
    }
}

static void send_int(int val)
{
    taskENTER_CRITICAL();
    printf("%d\n", val);
    taskEXIT_CRITICAL();
}

static void* interrupt_simulator(void* args)
{
    (void) args;

    while (true) {
        usleep(100000);

        pthread_kill(tid, SIGUSR1);
    }

    return NULL;
}

```

doppioandante, 2024-09-10:
From a quick glance, that won't cut it, but I'll have to think about it more thoroughly.
Since I've posted this issue, I've changed the architecture of my application getting rid of freertos queues, and using a lockless ring buffer to send data from the interrupt to the task. This allowed me to run the same code on the MCU and on the posix layer, although a bit inconvenient.
I think that, unless I'm missing something obvious, the POSIX port should look more similar to the Windows port (whose implementation is quite a bit more nuanced) in order for my use case to work.

aggarg, 2024-09-17:
> I think that, unless I'm missing something obvious, the POSIX port should look more similar to the Windows port (whose implementation is quite a bit more nuanced) in order for my use case to work.

We have not tried that. If you decide to write one, feel free to upstream.

> I'll reach back when I have time to do that.

I am closing this issue. Feel free to reopen or create a new one whenever you need anything.

closed 2024-09-17 by aggarg