FreeRTOS/FreeRTOS-Kernel #1077: [BUG] I used Posix gcc to compile on glibc 2.24 and gcc 6.3.0, but failed to compile the demo

maxiaogood, 2024-06-05:
**Describe the bug**
I used Posix gcc to compile on glibc 2.24 and gcc 6.3.0, but failed to compile the demo

**Target**
- Development board: [GCC_POSIX]
- Instruction Set Architecture: [x86]
- IDE and version: [glibc 2.24 and gcc 6.3.0]
- Toolchain and version: [glibc 2.24 and gcc 6.3.0]

**Host**
- Host OS: [linux (docker)]
- Version: [Linux version 6.5.0-35-generic (buildd@lcy02-amd64-079) (x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #35~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue May  7 09:00:52 UTC 2]

**To Reproduce**
- use project :FreeRTOS/Demo/Posix_GCC/CMakeLists.txt
- configure : default

**Expected behavior**
Can be compiled

**Screenshots**
[image]

**Additional context**
-- Configuring done
-- Generating done
-- Build files have been written to: /work/001_freertos/FreeRTOS/Demo/Posix_GCC/build
Consolidate compiler generated dependencies of target freertos_kernel_port
[  2%] Building C object FreeRTOS-Kernel/portable/CMakeFiles/freertos_kernel_port.dir/ThirdParty/GCC/Posix/port.c.o
In file included from /work/001_freertos/FreeRTOS/Source/portable/ThirdParty/GCC/Posix/port.c:62:0:
/usr/include/stdio.h:316:6: error: unknown type name '_IO_cookie_io_functions_t'
      _IO_cookie_io_functions_t __io_funcs) __THROW __wur;
      ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /work/001_freertos/FreeRTOS/Source/include/./FreeRTOS.h:58:0,
                 from /work/001_freertos/FreeRTOS/Source/portable/ThirdParty/GCC/Posix/port.c:75:
/work/001_freertos/FreeRTOS/Demo/Posix_GCC/./FreeRTOSConfig.h:178:33: fatal error: trcRecorder.h: No such file or directory
         #include "trcRecorder.h"
                                 ^
compilation terminated.
FreeRTOS-Kernel/portable/CMakeFiles/freertos_kernel_port.dir/build.make:75: recipe for target 'FreeRTOS-Kernel/portable/CMakeFiles/freertos_kernel_port.dir/ThirdParty/GCC/Posix/port.c.o' failed
make[2]: *** [FreeRTOS-Kernel/portable/CMakeFiles/freertos_kernel_port.dir/ThirdParty/GCC/Posix/port.c.o] Error 1
CMakeFiles/Makefile2:187: recipe for target 'FreeRTOS-Kernel/portable/CMakeFiles/freertos_kernel_port.dir/all' failed
make[1]: *** [FreeRTOS-Kernel/portable/CMakeFiles/freertos_kernel_port.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2

aggarg, 2024-06-05:
There seems to be 2 errors in the logs you shared: 

1.
```
In file included from /work/001_freertos/FreeRTOS/Source/portable/ThirdParty/GCC/Posix/port.c:62:0:
/usr/include/stdio.h:316:6: error: unknown type name '_IO_cookie_io_functions_t'
```

Does this go away if you change [this line](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/V11.1.0/portable/ThirdParty/GCC/Posix/port.c#L55) to:
```c
#define _GNU_SOURCE
```

2.
```
n file included from /work/001_freertos/FreeRTOS/Source/include/./FreeRTOS.h:58:0,
from /work/001_freertos/FreeRTOS/Source/portable/ThirdParty/GCC/Posix/port.c:75:
/work/001_freertos/FreeRTOS/Demo/Posix_GCC/./FreeRTOSConfig.h:178:33: fatal error: trcRecorder.h: No such file or directory
#include "trcRecorder.h"
^
```
Did you clone the submodules? You can do so with the following command:
```
git submodule update --init --recursive
```

closed 2024-06-10 by aggarg

maxiaogood, 2024-06-10:
1. This error is resolved using # define _GNU_SOURCE
In file included from /work/001_freertos/FreeRTOS/Source/portable/ThirdParty/GCC/Posix/port.c:62:0:
/usr/include/stdio.h:316:6: error: unknown type name '_IO_cookie_io_functions_t'

2.After using git submodule update -- init -- recursive, there is no second error left
n file included from /work/001_freertos/FreeRTOS/Source/include/./FreeRTOS.h:58:0,
from /work/001_freertos/FreeRTOS/Source/portable/ThirdParty/GCC/Posix/port.c:75:
/work/001_freertos/FreeRTOS/Demo/Posix_GCC/./FreeRTOSConfig.h:178:33: fatal error: trcRecorder.h: No such file or directory
#include "trcRecorder.h"

aggarg, 2024-06-10:
> 2.After using git submodule update -- init -- recursive, there is no second error left

So all your problems are solved, right?

maxiaogood, 2024-06-11:
yes

aggarg, 2024-06-11:
Thank you for confirming.