FreeRTOS/FreeRTOS-Kernel #1091: [BUG] When configUSE_PREEMPTION=0, removing own task does not switch to another task

koichimatsuno, 2024-06-12:
**Describe the bug**

In Windows, if the SuspendThread is called during thread execution, it may deadlock, so I configure configUSE_PREEMPTION=0 and use the taskYIELD.

When removing own task using the vTaskDelete in this environment, a symptom occurs that the task does not switch to another task after the task is removed.

In the vTaskDelete, the taskYIELD_WITHIN_API is called, but is the process of setting xSwitchRequired to TRUE in the xTaskIncrementTick  missing?

**Target**
- Windows10
- Visual Studio 2022

**Expected behavior**
- After removing own task, another task is executed.
- Applying the [FreeRTOS_v11.1.0.patch](https://github.com/user-attachments/files/15802143/FreeRTOS_v11.1.0.patch) to FreeRTOS_v11.1.0 worked as expected.

rawalexe, 2024-06-14:
Thank you for reaching out and pointing out the bug, I see you are also providing a patch, can you please create a PR against it so that we can go through the formal PR review and you will also get credit for it.

Best Regards,
AR

aggarg, 2024-08-29:
> In the vTaskDelete, the taskYIELD_WITHIN_API is called

The problem is that taskYIELD_WITHIN_API  does not get called for the Windows port and that is why the next task does not get scheduled. 

This [PR](https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/1132) fixes the issue by recording a yield in the portPRE_TASK_DELETE_HOOK.

Thank you for reporting this.

aggarg, 2024-08-29:
I am closing this issue as the PR has been merged. Feel free to reopen if you need anything from us.

closed 2024-08-29 by aggarg