FreeRTOS/FreeRTOS-Kernel #1114: [BUG] listGET_OWNER_OF_HEAD_ENTRY in fucntion xTaskResumeAll is optimized on xtensa platform

superroc, 2024-08-09:
**Describe the bug**
When more than 2 items exist in xPendingReadyList, pxTCB dose not update after the first loop, but retain the first value.

**Target**
- Development board: [SoC in developing procedure]
- Instruction Set Architecture: [tensilica hifi3z]
- IDE and version: [Xtensa Development Environment Version 10.1.11.3000]
- Toolchain and version: [RI-2022.9-win32 14.09 patch 20220930767596]

**Host**
- Host OS: [windows]
- Version: [windows 7]

**Screenshots**
this screenshot shows compile options which does not work well.
[image]

**Additional context**
the following is the compilation result of xTaskResumeAll:

78402998 <xTaskResumeAll>:
{
78402998:	004136               	entry	a1, 32
    configASSERT( uxSchedulerSuspended );
7840299b:	0020c0               	memw
7840299e:	b3fc599d394e              	{ l32r	a3, 78402684 (7821c3c0 <pxReadyTasksLists>); movi	a10, 179 }
784029a4:	7bf8457dbf4e              	{ l32r	a11, 784025a0 (78204b50 <uxTopUsedPriority+0x4>); addmi	a10, a10, 0x800 }
784029aa:	282322               	l32i	a2, a3, 160
784029ad:	42cc                	bnez.n	a2, 784029b5 <xTaskResumeAll+0x1d>
784029af:	0020f0               	nop
784029b2:	0234a5               	call8	78404cfc <vAssertCalled>

784029b5 <xTaskResumeAll+0x1d>:
        if( xSchedulerRunning != pdFALSE )
784029b5:	0020c0               	memw
784029b8:	272322               	l32i	a2, a3, 156
784029bb:	026216               	beqz	a2, 784029e5 <xTaskResumeAll+0x4d>
    uint32_t ps = XT_RSR_PS ();
784029be:	03e620               	rsr.ps	a2
784029c1:	f9f541               	l32r	a4, 78401198 (70f30 <_memmap_cacheattr_wb_base+0x2caee>)
    XT_WSR_PS ((ps & ~0xFU) | (level & 0xFU));
784029c4:	102240               	and	a2, a2, a4
784029c7:	02c222               	addi	a2, a2, 2
784029ca:	13e620               	wsr.ps	a2
    XT_RSYNC ();
784029cd:	002010               	rsync
            ( pxCurrentTCB->uxCriticalNesting )++;
784029d0:	0020c0               	memw
784029d3:	232322               	l32i	a2, a3, 140
784029d6:	102242               	l32i	a4, a2, 64
784029d9:	01c442               	addi	a4, a4, 1
784029dc:	106242               	s32i	a4, a2, 64
            if( pxCurrentTCB->uxCriticalNesting == 1 )
784029df:	0020c0               	memw
784029e2:	232322               	l32i	a2, a3, 140
        --uxSchedulerSuspended;
784029e5:	0020c0               	memw
784029e8:	282322               	l32i	a2, a3, 160
784029eb:	220b                	addi.n	a2, a2, -1
784029ed:	0020c0               	memw
784029f0:	286322               	s32i	a2, a3, 160
        if( uxSchedulerSuspended == ( UBaseType_t ) pdFALSE )
784029f3:	0020c0               	memw
784029f6:	4ac182022e            	{ movi.n	a2, 0; l32i	a4, a3, 160 }
784029fb:	348c                	beqz.n	a4, 78402a02 <xTaskResumeAll+0x6a>
    taskEXIT_CRITICAL();
784029fd:	ffe4e5               	call8	7840284c <vTaskExitCritical>
    return xAlreadyYielded;
78402a00:	f01d                	retw.n

[... 91 lines cut ...]
                            if( xTaskIncrementTick() != pdFALSE )
78402aee:	0020f0               	nop
78402af1:	002ea5               	call8	78402ddc <xTaskIncrementTick>
78402af4:	4a8c                	beqz.n	a10, 78402afc <xTaskResumeAll+0x164>
                                xYieldPending = pdTRUE;
78402af6:	0020c0               	memw
78402af9:	2d6352               	s32i	a5, a3, 180
                            --xPendedCounts;
78402afc:	440b                	addi.n	a4, a4, -1
                        } while( xPendedCounts > ( TickType_t ) 0U );
78402afe:	fec456               	bnez	a4, 78402aee <xTaskResumeAll+0x156>
                        xPendedTicks = 0;
78402b01:	0020c0               	memw
78402b04:	040c                	movi.n	a4, 0
78402b06:	306342               	s32i	a4, a3, 192
                if( xYieldPending != pdFALSE )
78402b09:	0020c0               	memw
78402b0c:	2d2332               	l32i	a3, a3, 180
78402b0f:	eea316               	beqz	a3, 784029fd <xTaskResumeAll+0x65>
                    taskYIELD_IF_USING_PREEMPTION();
78402b12:	fe8165               	call8	78401328 <vPortYield>
78402b15:	1ec3f702216e              	{ movi	a2, 1; j	784029fd <xTaskResumeAll+0x65> }

78402b1b <xTaskResumeAll+0x183>:
	...

tony-josi-aws, 2024-08-10:
@superroc 
Thanks for taking time to report this. 
Just wondering if you have checked if the same application code works well with lower optimization settings?

superroc, 2024-08-11:
Hi tony,

The same application code works well with lower optimization settings. "optimization level" is O0, "optimization for size" is None.

superroc, 2024-08-11:
This seems to be a compiler problem. The code works well after I insert a nop instruction following listGET_OWNER_OF_HEAD_ENTRY when optimization level is O2.
```
pxTCB = listGET_OWNER_OF_HEAD_ENTRY( ( &xPendingReadyList ) );
__asm__ __volatile__("nop" : : : "memory");
listREMOVE_ITEM( &( pxTCB->xEventListItem ) );
portMEMORY_BARRIER();
listREMOVE_ITEM( &( pxTCB->xStateListItem ) );
prvAddTaskToReadyList( pxTCB );
```

aggarg, 2024-08-11:
Can you check what if `portMEMORY_BARRIER` is defined in your portmacro.h?

superroc, 2024-08-11:
This macro is undefined in my project

superroc, 2024-08-11:
According you tips, I append defination
`#define portMEMORY_BARRIER()            __asm__ __volatile__("nop" : : : "memory")`
 in portmacro.h, and I checked the assembly file, it seems work well. Now it's midnight in my country and I don't have EVM board in my hand, I will test the change tomorrow morning. 
Thanks a lot!

superroc, 2024-08-12:
I have test this solution with different optimization level (O2 and O3) on my project and EVM board, it works well. Thank you very much for your advice.

Is it a already known problem? Should this patch be merged into portable/ThirdParty/XCC/Xtensa/portmacro.h?

aggarg, 2024-08-12:
> I have test this solution with different optimization level (O2 and O3) on my project and EVM board, it works well. 

Glad that it works for you.

>  Should this patch be merged into portable/ThirdParty/XCC/Xtensa/portmacro.h?

Yes, please raise a PR for it.

closed 2024-08-12 by superroc