FreeRTOS/FreeRTOS-Kernel #1115: [BUG] listGET_OWNER_OF_HEAD_ENTRY results in null ptr dereference when xTaskIncrementTick is called prior to xTaskCreate

JayToltTech, 2024-08-11:
**Describe the bug**

I am creating a STM32 (WB5M) FreeRTOS static-only application from scratch without using the CubeMX FreeRTOS libraries.  I have used CubeMX to generate a main() function to initialize the application.

I have commented out SVC_Handler() from stm32wbxx_it.c and added #define handlers per [compiles but does not run](https://www.freertos.org/Why-FreeRTOS/FAQs/Troubleshooting#the-application-i-created-compiles-but-does-not-run).

When I execute my code under the debugger, I get a hard fault after SystemClock_Config() is called which I have traced into port.c->SysTick_Handler() calling into xTaskIncrementTick() which then calls `listGET_OWNER_OF_HEAD_ENTRY( pxDelayedTaskList )` which dereferences a nullptr because `pxDelayedTaskList == nullptr`.

The behavior expectation of FreeRTOS seems to be that xTaskCreate... will be called prior to SysTick_Handler() because pxDelayedTaskList is initialized in `prvInitialiseTaskLists()` which is called by `prvInitialiseNewTask()`.  At least in my experience, this isn't true because I am calling the CubeMX generated functions that initialize the hardware and set up SysTick prior to any calls to xTaskCreate.

I don't see any guidance or documentation to this effect, and there is no 'initialize OS' function in FreeRTOS (similar say to OS_Init() in embOS) which initializes data structures and pointers early in the lifetime of the application.

**Target**
- Custom board based on the STM32WB5M (similar in design to P-Nucleo-WB55)
- Instruction Set Architecture: Cortex M4F
- IDE and version: VSCode/CMake/ninja
- Toolchain and version: arm-none-eabi-gcc 13.3.1 20240614

**Host**
- Host OS: MacOS
- Version: Sonoma 14.6.1

**To Reproduce**
Custom project based on CubeMX with main.c initializing SysTick prior to any calls to xTaskCreate

**Expected behavior**
xTaskIncrementTick / listGET_OWNER_OF_HEAD_ENTRY should either be tolerant of pxDelayedTaskList == nullptr or it should detect and call prvInitialiseTaskLists when needed or there should be some equivalent of OS_Init() which preinitializes dependent data structures prior to SysTick interrupt handling.

**Additional context**
[image]

[image]

[image]

[image]

``` c
/*
 * FreeRTOS Kernel V11.1.0
 * list.h
 * ...
*/

 * \page listGET_OWNER_OF_HEAD_ENTRY listGET_OWNER_OF_HEAD_ENTRY
 * \ingroup LinkedList
 */
#define listGET_OWNER_OF_HEAD_ENTRY( pxList )            ( ( &( ( pxList )->xListEnd ) )->pxNext->pvOwner )
```

aggarg, 2024-08-12:
> The behavior expectation of FreeRTOS seems to be that xTaskCreate... will be called prior to SysTick_Handler() because pxDelayedTaskList is initialized in prvInitialiseTaskLists() which is called by prvInitialiseNewTask().

The expectation is that the SysTick is not called until the scheduler is started. 

> I am calling the CubeMX generated functions that initialize the hardware and set up SysTick prior to any calls to xTaskCreate.

The CubeMX generated code is not supposed to setup SysTick as it is done by FreeRTOS. Most likely, you are using SysTick to drive ST HAL Tick too and that is why the generated code is setting up SysTick. Please change HAL Tick to use some other timer. You can do so in CubeMX UI by going to `System Core --> SYS --> Timebase Source`:

[image]

JayToltTech, 2024-08-12:
I did discover some docs on SYS Timebase / SysTick and made this change after submitting this bug.  Have verified that the bug does not reproduce when SYS Timebase Source is changed from SysTick to another timer like TIM1 in CubeMX.

I was curious if there was another path to get to nullptr deref, so I commented out all task and timer inits from my code leaving just cubemx_main() and vTaskStartScheduler(), but this did *not* result in an uninitialized pointer deref in listGET_OWNER_OF_HEAD_ENTRY.

The code does look fragile to me, but I wasn't able to find a 'valid' path to uncover the problem.

closed 2024-08-12 by JayToltTech