FreeRTOS/FreeRTOS-Kernel #1120: Real time task statistics in the uxTaskGetSystemState()

RomkaE, 2024-08-15:
**Problem:** Statistics collection API provide only AVERAGE statistics since kernel startup and don't calculate real time statistics.

Hi guys,
I would like to start a discussion on this issue and make changes to the Kernel to expand the API for collecting real time task statistics.

**Example 1:** the task was blocked for 10 minutes, and on event woke up for 1 second and worked for 500ms. The current implementation calculates statistics as 100*0.5/10*60 = 0.083(3)%. In integer representation it is equal to 0. But real time statistics for the last second is equal 50%, or for the last 10 seconds is equal 5%.

**Example 2:** one task is created only for initialization. For 5 seconds the task used 100% of CPU time, after initialization the task is deleted. In this case the sum of statistics of the remaining tasks (including the IDLE task) is less than 100%.

I found a couple of topics with similar discussions but in a different context:
[Issue #307](https://github.com/FreeRTOS/FreeRTOS-Kernel/issues/307)
[Forum: How to reset vTaskGetRunTimeStats?](https://forums.freertos.org/t/how-to-reset-vtaskgetruntimestats/7861)

I've been using task statistics collection in every my projects with FreeRTOS to analyze and optimize CPU performance and memory consumption. I use uxTaskGetSystemState() and my own implementation of formatted output via UART.
To calculate real time statistics I use the following changes: [commit fe0a6c0](https://github.com/FreeRTOS/FreeRTOS-Kernel/commit/fe0a6c0a393648f9529769dc105483bcb7280213)
But these changes cause other APIs to not work correctly. And I understand that I can't create a pull request with such changes.

And the last point as an argument in favor of real time statistics: couple screenshots from production project with real time statistics
[image]
[image]

In the screenshots, the statistics update time is 1 sec. The screenshots show peak loads on the processor. With average statistics calculations, the peak load is not visible.

cookpate, 2024-08-15:
Thanks for the insight!

amazonKamath, 2024-08-16:
@RomkaE Thanks for the opening the issue here, but in order to get community feedback and involvement, we would request you to create this as a forum post and close the github enhancement issue here. Thank you.

RomkaE, 2024-08-16:
@amazonKamath ok. Thanks.

closed 2024-08-16 by RomkaE