FreeRTOS/FreeRTOS-Kernel #1121: Port IA32_flat assembly code no longer compiles correctly.

Ryzee119, 2024-08-16:
**Describe the bug**
I have bisected to this commit https://github.com/FreeRTOS/FreeRTOS-Kernel/commit/5fb9b50da8f1e02fa0c81ee43d7961c479ef5f12 which is simply a code formatting PR, however this must break some syntax within the assembly code

The problem appears to lie within [portable/GCC/IA32_flat/ISR_Support.h](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/portable/GCC/IA32_flat/ISR_Support.h) which is included by `portASM.S`.

Compiling with `i686-elf-gcc` on current master I get the following. Going back one further commit works as expected.

```
lib/freertos_kernel/portable/GCC/IA32_flat/portASM.S: Assembler messages:
lib/freertos_kernel/portable/GCC/IA32_flat/portASM.S:179: Warning: no instruction mnemonic suffix given and no register operands; using default for `add'
lib/freertos_kernel/portable/GCC/IA32_flat/portASM.S:196: Warning: no instruction mnemonic suffix given and no register operands; using default for `sub'
lib/freertos_kernel/portable/GCC/IA32_flat/portASM.S:231: Error: unknown pseudo-op: `.'
lib/freertos_kernel/portable/GCC/IA32_flat/portASM.S:231: Error: no such instruction: `if 1==1'
lib/freertos_kernel/portable/GCC/IA32_flat/portASM.S:231: Error: invalid character '(' in mnemonic
lib/freertos_kernel/portable/GCC/IA32_flat/portASM.S:231: Warning: no instruction mnemonic suffix given and no register operands; using default for `add'
lib/freertos_kernel/portable/GCC/IA32_flat/portASM.S:254: Warning: no instruction mnemonic suffix given and no register operands; using default for `sub'
lib/freertos_kernel/portable/GCC/IA32_flat/portASM.S:254: Error: junk `movl( % eax)' after register
lib/freertos_kernel/portable/GCC/IA32_flat/portASM.S:254: Error: unknown pseudo-op: `.'
lib/freertos_kernel/portable/GCC/IA32_flat/portASM.S:254: Error: no such instruction: `if 1==1'
lib/freertos_kernel/portable/GCC/IA32_flat/portASM.S:254: Error: invalid character '(' in mnemonic
lib/freertos_kernel/portable/GCC/IA32_flat/portASM.S:254: Error: ".endif" without ".if"
lib/freertos_kernel/portable/GCC/IA32_flat/portASM.S:258: Error: ".endif" without ".if"
```

**Target**
- i686-elf-gcc
- custom x86 target

**Host**
- Host OS: Windows 11 (WSL2 Ubuntu)

**To Reproduce**
```
i686-elf-gcc -Wa,--gdwarf2 -Wa,-march=pentium -c -o portASM.o portable/GCC/IA32_flat/portASM.S
```

**Expected behavior**
Should compile.  It works as expected if I go one commit back. ie to https://github.com/FreeRTOS/FreeRTOS-Kernel/commit/01820d3ed93c735e5e8460a282971a7725fae44c

**Screenshots**
N/A

kar-rahul-aws, 2024-08-16:
Hi @Ryzee119 
Thank you for pointing out the issue. We were able to reproduce the issue you mentioned, and raised a [patch](https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/1122) to fix the same. Could you please give this a try to see if it fixes the issue?

Thanks

Ryzee119, 2024-08-17:
@kar-rahul-aws thank you for the quick fix. I can confirm that this fixes the issue.

kar-rahul-aws, 2024-08-19:
@Ryzee119 Thank you for reporting back.

closed 2024-08-19 by kar-rahul-aws