FreeRTOS/FreeRTOS-Kernel #1134: [Feature Request] Dynamic Prioritization

svenbieg, 2024-09-01:
The problem is to define the priority on a task-level. Any task can call a critical function, so the priority has to be raised there.

My solution is a task-lock that works with a mutex. The task is marked exclusive, so interrupts are kept disabled after the task-switch until the task-lock is released.

You can take a look at the heap for example. It is used by every task and is a critical function. I'm using a task-lock there.

Best regards,

Sven Bieg

aggarg, 2024-09-03:
Please create a post in the [FreeRTOS Forums](https://forums.freertos.org/) explaining your feature request in more detail, possibly with sequence diagrams to help us understand your request.

svenbieg, 2024-09-03:
This time i'm sorry. You can find the detais here:

https://github.com/svenbieg/Core

I'm happy to see You again, looking forward to my Christmas holidays. My kernel took me two weeks, it's really easy with know-how.
Your vTaskSuspendAll is too expensive, You need a mutex there that can be locked exclusive. 

Good luck, to the whole FreeRTOS-team!

I'm really sorry i don't have time at the moment,

Sven Bieg

aggarg, 2024-09-03:
Thank you for sharing your implementation. I assume you do not need anything from FreeRTOS for your implementation as your implementation does not depend on FreeRTOS.

closed 2024-09-03 by n9wxu

svenbieg, 2024-09-04:
Semaphore sounds like a german joke, "we are planing". Mutex meaning is clear mutual exclusive.

svenbieg, 2024-09-08:
What about mutex_lock, mutex_lock_shared and mutex_lock_up?

svenbieg, 2024-09-10:
You are right, exclusive mutual exclusion sounds funny, but it is exactly what You are missing.
If we have 100 tasks, access the heap, a high priority task is waiting, and the system timer interrupts, the high priority task has to wait for 100 tasks.
What about mutex_lock, mutex_lock_shared and mutex_lock_up?

svenbieg, 2024-09-10:
FIFO task management

svenbieg, 2024-09-15:
I guess this example is easier to read:

https://github.com/svenbieg/Blink

Happy Weekend!

svenbieg, 2025-02-02:
https://github.com/svenbieg/Scheduler