FreeRTOS/FreeRTOS-Kernel #1155: [Feature Request] Adding matching branch names to the released Tags

finger42, 2024-10-11:
**Is your feature request related to a problem? Please describe.**

I am using this Git repo as submodule in my projekt. 
Now I want to get the released version to access also the SBOM file, like:
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/V11.1.0/sbom.spdx .

But this Tag does not belong to any Branch. 

"This commit does not belong to any branch on this repository, and may belong to a fork outside of the repository."

**Describe the solution you'd like**
The release Action on Github should create a branch that matches to the specific Tag.

rawalexe, 2024-10-11:
Hello @finger42,
The release process it done so on purpose. You may add a submodule pointer to your tagged version by simply cloning the repo `cd` into the repo and then just checkout the tagged release,
`git checkout V11.1.0`
Just make sure that the `.gitmodules` is also updated correctly afterwards. Hope that helps. Please let us if you have more questions.

kar-rahul-aws, 2024-10-31:
Hi @finger42 
Do you have any more questions or concerns regarding the above explanation ?

finger42, 2024-11-04:
Hello @kar-rahul-aws 
Sorry for my late response.

I am aware how to change or checkout the Tag of an specific version. Also It is possible to checkout the given hash.

My Setup is more complex than I described it in my initial opening of this Issue.

I am using Gerrit to host my Git repos. And with it, the subscription feature. 
( https://gerrit-review.googlesource.com/Documentation/user-submodules.html )

With this feature I build something like a dependabot. My FreeRTOS repo contains just one special branch, e.g. my-main.
And all of my parent repos which using the FreeRTOS as a submodule are subscriped to this branch (my-main).

If I like to update all Git repos at once I  merge the given Branch like ( V11.1.0 ) to my-main.

But on Gerrit it is hard to do this if the commit does not belong to an branch. This is indicated by the msg:

"This commit does not belong to any branch on this repository, and may belong to a fork outside of the repository."

I can workaround this problem on my Gerrit instance by crafting a new branch, or force the merge to my-main branch. 

It would be much nicer if the generated commits in FreeRTOS are based on an branch. So that the Github release action creates a branch and put all of the new commits inside this new branch. So that the branch name and the given Tag are the same.

xuelix, 2024-11-26:
Hi, @finger42 
From the link you had above about Gerrit, it might accept any refspec.  Just wondering if you have tried ref/tags/V11.1.0 instead of ref/heads/xx.  I don't know if Gerrit submodules would accept that or not.

The reason we do not create a separate branch for a release is to simplify the maintenance.  You can always create branch off the release tag like you suggested in your workaround.

Thanks

finger42, 2024-11-27:
Hello, @xuelix 

Thank you for the feedback. 

From my point of view it is ok to close this issue.

closed 2024-11-27 by finger42