FreeRTOS/FreeRTOS-Kernel #1159: [DOC]CONFUSION -  configCHECK_FOR_STACK_OVERFLOW verses portHAS_STACK_OVERFLOW_CHECKING

duaneellissd, 2024-10-16:
When I read these - i see: config CHECK STACK OVERFLOW - and  port STACK OVER FLOW CHECK

These are very similar and seem to be very unrelated - as if both are intended to be and do the exact same thing.

kar-rahul-aws, 2024-10-17:
Hi @duaneellissd 
`configCHECK_FOR_STACK_OVERFLOW` enables software stack overflow checking and it is implemented for all ports. The application writer can set it to 1 or 2 in their FreeRTOSConfig.h to enable software stack-overflow checking.

`portHAS_STACK_OVERFLOW_CHECKING` enables hardware stack-overflow checking and is only implemented for ports where the hardware supports it (for example, ARMv8-M supports hardware stack-overflow checking using stack limit registers). This macro is set by the port itself and the application is not supposed to set it in FreeRTOSConfig.h.

ActoryOu, 2024-10-31:
Hi @duaneellissd,
I hope @kar-rahul-aws's explanation has been helpful in addressing your concerns.

We want to ensure that all your questions have been answered satisfactorily. Is there any aspect of the explanation that you'd like us to clarify further, or do you have any additional questions?

Thank you.

duaneellissd, 2024-10-31:
perhaps the name portHAS_STACK_OVERFLOW would be better named portHAS_HW_STACK_OVERFLOW

ActoryOu, 2024-11-02:
The "port" prefix is indeed used to indicate hardware-related elements. All definitions directly associated with hardware components carry this prefix.

ActoryOu, 2024-11-05:
As there have been no further concerns raised, we'll be closing this issue. Please feel free to reopen it if you have any new points to discuss.

closed 2024-11-05 by ActoryOu