FreeRTOS/FreeRTOS-Kernel #1165: [BUG] Inconsistent parameter name

swaldhoer, 2024-10-25:
**Describe the bug**
In FreeRTOS version V10.5.0 a parameter in function `MPU_vTimerSetReloadMode` was changed from `UBaseType_t uxAutoReload` to `BaseType_t xAutoReload` : https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/7215c89aa81501f60af8df68dfa99fc2c118a293/History.txt#L414-L417
But there are still old namings used which not all compilers and coding standards are happy about.

**Target**
- Development board: n. a.
- Instruction Set Architecture: n. a.
- IDE and version: n. a.
- Toolchain and version: n. a.

**Host**
- Host OS: n. a.
- Version: n. a.

**To Reproduce**
See https://github.com/search?q=repo%3AFreeRTOS%2FFreeRTOS-Kernel%20uxAutoReload%20&type=code

**Expected behavior**
Parameter name is in all declarations and definitions `xAutoReload`.

archigup, 2024-10-25:
Hey, this does seem like a bug.

Feel free to open a PR if you would like, else we can do so.

kstribrnAmzn, 2024-10-30:
Went ahead and made the fix - https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/1166

kar-rahul-aws, 2024-10-31:
Hi @swaldhoer 
The PR #1166 has been approved and merged. Does it fix the issue you reported?

kstribrnAmzn, 2024-10-31:
Closing since the issue is fixed.

swaldhoer, 2024-11-01:
No, it's only partially fixed, see https://github.com/search?q=repo%3AFreeRTOS%2FFreeRTOS-Kernel%20uxAutoReload&type=code

kstribrnAmzn, 2024-11-01:
Thanks for the search! I see I forgot at least a couple which need to be fixed.

swaldhoer, 2024-11-01:
Yes, but for the remaining `uxAutoReload` it's not just about the name, but also the type.

kstribrnAmzn, 2024-11-01:
Agreed! I'm going to revert my commit, figure out how we got the mismatched types across the mpu wrapper v1 and v2, and then will come up with a fix commit.

kstribrnAmzn, 2024-11-01:
The type mismatches were straightforward so I've created a PR to apply the type mismatch fix - https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/1181

kstribrnAmzn, 2024-11-01:
All problems should be fixed now - let me know if you still find any other issues.

swaldhoer, 2024-11-01:
Thanks, will have a look, when updating to the latest version, but I think it should be good!

Small question: The changelog has not been updated on purpose? Is it only updated on new releases or how does it work?

kstribrnAmzn, 2024-11-01:
I took a look at the [history.txt](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/History.txt) log but it only seems to document changes between releases. We will mention this fix between as part of our next release.

closed 2024-11-01 by swaldhoer