FreeRTOS/FreeRTOS-Kernel #1167: [DOC]: fix spelling typos

rahmanih, 2024-10-31:
Hi,

The codespell tool has detected some spelling mistakes. fixed in the attached patch.

[fix_spelling_typos.TGZ](https://github.com/user-attachments/files/17585287/fix_spelling_typos.TGZ)

regards
haithem.

ActoryOu, 2024-10-31:
Hi @rahmanih,
Thanks for your help! I applied the patch and created PR.
Note that I changed a bit for some terminologies or MACRO naming like below.

portable/GCC/ARM7_AT91SAM7S/lib_AT91SAM7X256.h:
~~~diff 
-  /** \brief Cfg MORE Register according to the main osc startup time */
+  /** \brief Cfg MOR Register according to the main osc startup time */
~~~

rahmanih, 2024-10-31:
hi @ActoryOu,
thanks for the quick update.
just one more point,there are some typos reported in the History.txt file that I didn't share in the previous patch, to avoid any confusion it may introduce. may be the fix can be applied once the file is getting updated for a new release.

up to you to decide, below the typos revealed by codespell.

````
History.txt:531: Symetric ==> Symmetric
History.txt:2018: revered ==> reversed
History.txt:3177: chaned ==> changed, chained
History.txt:3311: psuedo ==> pseudo
````

regards
Haithem.

ActoryOu, 2024-11-01:
Hi @rahmanih,
Thanks for suggestion. Updated PR.

kstribrnAmzn, 2024-11-01:
I've approved and merged the PR. Feel free to reopen if other typos still exist.

closed 2024-11-01 by kstribrnAmzn