FreeRTOS/FreeRTOS-Kernel #1172: [BUG] rp2040 issue debugging SMP

0x3333, 2024-11-01:
**Describe the bug**
In rp2040, if running FreeRTOS with SMP enabled, we can't debug because the processor got stuck after launching core1. Adding a `multicore_reset_core1` right before `multicore_launch_core1` fix the issue.

https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/8225a7f554a1c7c76f6f179bac37d32e47f40125/portable/ThirdParty/GCC/RP2040/port.c#L374-L385

**Target**
- Development board: Raspberry Pi Pico
- Instruction Set Architecture: ARM
- IDE and version: VSCode 1.95.1
- Toolchain and version: arm-none-eabi-gcc 13.3.Rel1

**Host**
- Host OS: MacOS
- Version: Sonoma 14.6.1

**Additional context**
I have a BlackMagic Debug probe and a Pico Debug Probe, both have the same issue.

ActoryOu, 2024-11-01:
Hi @0x3333,
Thank you for reporting this to us. To add more content, here is an [external discussion](https://forums.raspberrypi.com/viewtopic.php?t=365688) about this.

Thank you.

ActoryOu, 2024-11-01:
Hi @0x3333,
Could you help take a look on the #1174 to check if that solves your problem?

Thank you.

0x3333, 2024-11-01:
> Hi @0x3333, Thank you for reporting this to us. To add more content, here is an [external discussion](https://forums.raspberrypi.com/viewtopic.php?t=365688) about this.
> 
> Thank you.

Great, here are some context others:

https://forums.raspberrypi.com/viewtopic.php?p=2096741#p2096741

https://forums.raspberrypi.com/viewtopic.php?p=1894091#p1896571

https://forums.raspberrypi.com/viewtopic.php?p=1819358#p1816148

0x3333, 2024-11-04:
> In my tests, setting `timer_hw->dbgpause = 0;` before any timer related call in my code, the debugger works as expected, without the `multicore_reset_core1()` call.

Just an additional info. I encountered some situations where `timer_hw->dbgpause = 0;` doesn't work. However, `multicore_reset_core1()` always worked. I still need to dig down a little more, but the fix in the PR looks correct to me.

closed 2024-11-04 by aggarg