FreeRTOS/FreeRTOS-Kernel #1173: Inconsistent use of configASSERT in kernel functions

PwnVerse, 2024-11-01:
I was browsing through FreeRTOS's kernel code and wondered why there is inconsistent usage of configASSERTs in the many parts of the kernel code. For instance, in function `pvTaskGetThreadLocalStoragePointer`, the `pxTCB` is queried and directly referenced without checking if it exists or not [here](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/tasks.c#L5986)

```c
void * pvTaskGetThreadLocalStoragePointer( TaskHandle_t xTaskToQuery,
                                               BaseType_t xIndex )
    {
        if( ( xIndex >= 0 ) &&
            ( xIndex < ( BaseType_t ) configNUM_THREAD_LOCAL_STORAGE_POINTERS ) )
        {
            pxTCB = prvGetTCBFromHandle( xTaskToQuery );
            pvReturn = pxTCB->pvThreadLocalStoragePointers[ xIndex ];
        }
        else
        {
            pvReturn = NULL;
        }

        return pvReturn;
    }

```

But the same `pxTCB` is guarded by a `configASSERT` in other functions where it is [queried](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/tasks.c#L4197) - 

```c
char * pcTaskGetName( TaskHandle_t xTaskToQuery )
{
    TCB_t * pxTCB;

    traceENTER_pcTaskGetName( xTaskToQuery );

    /* If null is passed in here then the name of the calling task is being
     * queried. */
    pxTCB = prvGetTCBFromHandle( xTaskToQuery );
    configASSERT( pxTCB );

    traceRETURN_pcTaskGetName( &( pxTCB->pcTaskName[ 0 ] ) );

    return &( pxTCB->pcTaskName[ 0 ] );
}
```

Is there a reason for using `configASSERT` in one function and not using it in another? I might be entirely wrong in my judgement. It will be very helpful to know the developer's insight as to why such a choice is made. (PS : I reported this since my fuzzer crashed complaining about NULL dereferences in the `pvTaskGetThreadLocalStoragePointer`)

kar-rahul-aws, 2024-11-01:
Hi @PwnVerse 
Thank you for pointing out the issue. I have raised a PR #1177  for adding the assert check for NULL TCB handle . Let me know if that addresses your concern.

PwnVerse, 2024-11-01:
Thanks alot for the acknowledgement. Yes, it does resolve my concern.

ActoryOu, 2024-11-02:
Closing issue since #1177 has been merged.

closed 2024-11-02 by ActoryOu