FreeRTOS/FreeRTOS-Kernel #1184: [BUG] error: 'sio_hw' undeclared. Possible missing #include in RP2040 port.c

JLKenyon, 2024-11-04:
**Describe the bug**
I got the following error while trying to build an RP2040 hello world app:
```
/home/jlkenyon/Projects/SomeFreeRTOSExamples/submodules/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/port.c:1091:17: error: 'sio_hw' undeclared (first use in this function); did you mean 'scb_hw'?
 1091 |                 sio_hw->fifo_wr = 0;
      |                 ^~~~~~
      |                 scb_hw      
```                                  

It looks like the sio_hw object is defined in hardware/structs/sio.h in the pico-sdk, and the FreeRTOS RP2040/port.c references the clocks.h and exception.h from the pico-sdk already.

sio_hw definition:
https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2040/hardware_structs/include/hardware/structs/sio.h#L196

Proposed fix:
```diff
diff --git a/portable/ThirdParty/GCC/RP2040/port.c b/portable/ThirdParty/GCC/RP2040/port.c
index 6c5d0e513..7c2041993 100644
--- a/portable/ThirdParty/GCC/RP2040/port.c
+++ b/portable/ThirdParty/GCC/RP2040/port.c
@@ -36,6 +36,7 @@
 #include "rp2040_config.h"
 #include "hardware/clocks.h"
 #include "hardware/exception.h"
+#include "hardware/structs/sio.h"
 
 /*
  * LIB_PICO_MULTICORE == 1, if we are linked with pico_multicore (note that
```

**Target**
- Development board: Adafruit Feather RP2040
- Instruction Set Architecture: ARM Cortex M0+
- IDE and version: Vim/Ubuntu
- Toolchain and version: arm-none-eabi-gcc (15:13.2.rel1-2) 13.2.1 20231009

**Host**
- Host OS: Ubuntu 
- Version: 24.04

**To Reproduce**
- I am working on sharing the project in question to github as a separate project.  If will add a comment referencing this project if needed (and if I get it shared in a timely manner)

**Expected behavior**
Build RP2040 example without "undeclared" errors

**Screenshots**
N/A

**Additional context**
I am having trouble getting CMake to behave in a way that makes sense, so I am unsure if I setting up my project correctly.  I will share this information later if I can get my code posted for reference.

kar-rahul-aws, 2024-11-04:
Hi @JLKenyon 
We have few SMP [demos ](https://github.com/FreeRTOS/FreeRTOS-Community-Supported-Demos/tree/main/CORTEX_M0%2B_RP2040) using the Cortex M0+_RP2040 board . I tried building the Standard demo and was unable to reproduce the issue you mention. 

Host OS: Ubuntu
Version: 24.04

Can you take a look, if you are able to reproduce the issue with this demo and use the CMakeLists in this project as a reference .

JLKenyon, 2024-11-04:
I will take a look.  The demos linked above built, but I will need some cycles to compare the two setups.

kar-rahul-aws, 2024-11-04:
Sure, thank you for checking this out. Also, please help in sharing the project in which you are facing those issues.
Thanks.

kar-rahul-aws, 2024-11-08:
Hi @JLKenyon 
Did you get a chance to compare the 2 setups and see the working of the demos ?

JLKenyon, 2024-11-08:
Not yet. I am ok closing this ticket for the time being, no need to keep an idle ticket open IMO. If and when I can prove the bug is real with a minimal reproduction, I can open a new bug, and link to this one for reference.

JLKenyon, 2024-11-10:
On further investigation, I think the problem is better fixed in pico-sdk, not in FreeRTOS. Closing on this basis.

More specifically, there are two header files provided by pico-sdk:
`pico-sdk/src/rp2_common/pico_multicore/include/pico/multicore.h`
`pico-sdk/src/host/pico_multicore/include/pico/multicore.h`

The former includes the header file of interest: `hardware/structs/sio.h`
The later does not, and so is likely where the problem is. I am going to follow that lead to the pico-sdk repo.

Also, thanks for the links to the demos above, I was only able to understand this because of those examples. I should have found those examples on my own, so thanks for your patience as well.

closed 2024-11-10 by JLKenyon