FreeRTOS/FreeRTOS-Kernel #1200: [DOC] Download URL in sbom.spdx is not exactly the same than the tag URL

finger42, 2024-11-29:
**Describe the issue**
In the generated SBOM file:
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/V11.1.0/sbom.spdx#L15

is the URL: https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/v11.1.0
which leads to 404.
Reason is that the 'v' should be written upper case 'V'

I dont know why the URL is case sensitive?

So the URL: 
https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/V11.1.0
is working.

From my search the sbom.spdx file is generated by:
https://github.com/FreeRTOS/CI-CD-Github-Actions/blob/main/sbom-generator/scan_dir.py#L35
Where the manifest file gets parsed and generates the URL.

The manifest file:
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/V11.1.0/manifest.yml
contains the URL with lower 'v'

So the Tag URL and the manifest file version differs.

kstribrnAmzn, 2024-11-29:
What a silly thing to get wrong - thank you for catching this! It looks like the URL is case sensitive as it needs to match the tag exactly. Our release tags use a capitalized V.

For now, I'll raise a PR to update the main branch to use the capitalized V. I'll also start a discussion with the team to see if we want to release a patch version of FreeRTOS 11.1 to correct this URL.

kstribrnAmzn, 2024-11-29:
Fix - https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/1201

finger42, 2024-11-30:
Hello @kstribrnAmzn 

Thank you for fixing this.

From my point of view I can wait for the next release, like 11.2 or 12.x

Close the Issue ?

finger42, 2024-12-02:
Thx for your help !

closed 2024-12-02 by finger42